FreeCAD/FreeCAD #18130: placement dialog - unexpected imprecision introduced by rounding of values in QuantitySpinBox

pwolny, 2024-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When developing a patch allowing for manipulation of objects with spacemouse/spacenavd devices ( #18098 ) I have noticed that when I rotate object in small increments in one axis (lets say X) by 90deg then rotating it again by 90deg in perpendicular axis (Y) I can see a rotation on an untouched axis (Z).
It is easily noticeable with a custom 6dof controller that allows for precise control of rotation on each axis.
This can be mitigated by increasing the number of decimal places in general preferences.

Issue is triggered when storing values in QuantitySpinBox by using in code:
`ui->xAxis->setValue(new_dir_vect.x);` etc.
after those calls readout of value with:
`ui->xPos->value().getValue();`
returns a rounded value (with applied decimal places setting).

I think I found in the past a similar issue (I think it was resolved in some way) in another dialog but I lost the link and I can not find it again.

Should values stored in QuantitySpinBox, when set from code (with setValue), be stored internally with full precision and without rounding? 
Maybe only display and input in QuantitySpinBox should be rounded?
Or am I just using the QuantitySpinBox wrong?

### Full version info

```shell
issue seen on following versions of freecad (with applied patch from PR #18098 ):
[code]
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39290 (Git)
Build type: Unknown
Branch: spacenav-move-obj
Hash: 0a2e78be3872e92f9aa3a280d35ecff469d9fc42
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.61
  * CurvedShapes 1.0.5
  * DynamicData 2.59.0
  * Pyramids-and-Polyhedrons
  * Plot 2022.4.17
  * Assembly4 0.50.6
  * parts_library
  * SelectorToolbar
  * pcb 6.2023.0
  * Manipulator 1.5.0
  * toSketch 1.0.1
  * ThreadProfile 1.89.0
  * slic3r-tools
  * fasteners 0.4.74
  * lattice2 1.0.0
  * sheetmetal 0.5.6
  * kicadStepUpMod 10.22.1
  * ExplodedAssembly
  * MnesarcoUtils 0.2.5
  * CubeMenu
  * Estimate 0.1.2
  * freecad.gears 1.1.0
  * Silk 0.1.5
  * offline-documentation 1.0.0-alpha
  * QuickMeasure 2022.10.28
  * dodo 1.0.0
  * Defeaturing 1.2.1
[/code]

also on:
[code]
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.2.
Build type: Unknown
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.61
  * CurvedShapes 1.0.5
  * DynamicData 2.59.0
  * Pyramids-and-Polyhedrons
  * Plot 2022.4.17
  * Assembly4 0.50.6
  * parts_library
  * SelectorToolbar
  * pcb 6.2023.0
  * Manipulator 1.5.0
  * toSketch 1.0.1
  * ThreadProfile 1.89.0
  * slic3r-tools
  * fasteners 0.4.74
  * lattice2 1.0.0
  * sheetmetal 0.5.6
  * kicadStepUpMod 10.22.1
  * ExplodedAssembly
  * MnesarcoUtils 0.2.5
  * CubeMenu
  * Estimate 0.1.2
  * freecad.gears 1.1.0
  * Silk 0.1.5
  * offline-documentation 1.0.0-alpha
  * QuickMeasure 2022.10.28
  * dodo 1.0.0
  * Defeaturing 1.2.1
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-25:
@kadet1090 might be interesting

pwolny, 2024-11-27:
Maybe this example would be useful, when using patch from PR #18098.
With eventFilter-C and eventFilter-D printf statements uncommented it basically spits out values received from spacenavd device used be event filter, written into QuantitySpinBox (eventFilter-C line) and read out of QuantitySpinBox (eventFilter-D line) .

When rotating first by 10deg in Z-axis and then again by 10deg in Y-axis, I can get something like this (with 3 decimal places precision):
```
eventFilter-C	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.000000,0.000000,1.000000)	(RotAngle)=(0.000000)
eventFilter-D	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.000000,0.000000,1.000000)	(RotAngle)=(0.000000)

Spaceball Placement Event	(x,y,z)=(0.000000,0.000000,0.000000)	(Rx,Ry,Rz)=(0.000000,0.000000,10.000000);
eventFilter-C	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.000000,0.000000,1.000000)	(RotAngle)=(10.000000)
eventFilter-D	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.000000,0.000000,1.000000)	(RotAngle)=(10.000000)

Spaceball Placement Event	(x,y,z)=(0.000000,0.000000,0.000000)	(Rx,Ry,Rz)=(0.000000,10.000000,0.000000);
eventFilter-C	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.061746,0.705758,0.705758)	(RotAngle)=(14.133149)
eventFilter-D	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.062000,0.706000,0.706000)	(RotAngle)=(14.133000)
```
When using freecad with this low precision display setting I see cumulative errors when iterating over rounded values by rotating the object...

And with 16 decimal places precision when rotating first by 10deg in Z-axis and then again by 10deg in Y-axis, I can mitigate the problem and get something like this:
```
eventFilter-C	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.000000,0.000000,1.000000)	(RotAngle)=(0.000000)
eventFilter-D	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.000000,0.000000,1.000000)	(RotAngle)=(0.000000)

Spaceball Placement Event	(x,y,z)=(0.000000,0.000000,0.000000)	(Rx,Ry,Rz)=(0.000000,0.000000,10.000000);
eventFilter-C	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.000000,0.000000,1.000000)	(RotAngle)=(10.000000)
eventFilter-D	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.000000,0.000000,1.000000)	(RotAngle)=(10.000000)

Spaceball Placement Event	(x,y,z)=(0.000000,0.000000,0.000000)	(Rx,Ry,Rz)=(0.000000,10.000000,0.000000);
eventFilter-C	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.061746,0.705758,0.705758)	(RotAngle)=(14.133149)
eventFilter-D	(x,y,z)=(0.000000,0.000000,0.000000)	(RotXYZ)=(0.061746,0.705758,0.705758)	(RotAngle)=(14.133149)
```