FreeCAD/FreeCAD #18134: Appimage v1.0.0 did not work

yahbluez, 2024-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This Appimage:

https://github.com/FreeCAD/FreeCAD/releases/download/1.0.0/FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage

has only a size of 649 MB compared to ~ 800 MB a working one has.

It crashed every time with this message:
"execv error: No such file or directory"

### Full version info

```shell
Freecad 1.0.0-conda Linux x86_64 Appimage
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-26:
[image]

oursland, 2024-11-26:
@yahbluez Are you using AppImageLauncher, by chance?

tarman3, 2024-11-26:
No, AppImageLauncher does not installed.

yahbluez, 2024-11-26:
@oursland i try on manjaro linux and did not change anything. The older version like the RC2 work, the snap version also works. Starting with the launcher or from the command line booth do not work.

tarman3, 2024-11-26:
I think this has already been mentioned somewhere.
Try to uninstall AppImageLauncher.

yahbluez, 2024-11-26:
@tarman3 
[image]

tarman3, 2024-11-26:
show output
```
ls -l
```

tarman3, 2024-11-26:
To not distract system you can try to boot in live mode and try to run freecad without applauncher