FreeCAD/FreeCAD #18137: New CAD simulator shows only a portion of the model

plueschpruem, 2024-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Might be related to this: https://github.com/FreeCAD/FreeCAD/issues/14348

When I open the new simulator and press "Play", the window opens on the built in screen and shows only like 1/4 of the project.
[image]
When I push the window to my external screen (a BenQ with 2560×1440px) and resize / press play, the model resizes / shifts correctly. The rest of the simulator works as intended.
[image]

Disconnecting the external display, changing the internal resolution and restarting FreeCAD don't seem to change anything.

### Full version info

```shell
FreeCAD 1.0 (but I noticed this in the weekly betas too)
MacOS 13.7 (22H123) on i9 iMac 2019
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-26:
Please add full about info and an example file.
@shaise FYI

shaise, 2024-11-26:
1. Please attach the file used. (or a sample file)
2. when you resize the window back to original size, does it resize the model correctly, or does it shows the same issue?
3. Does the mouse pan/resize/rotate of the model works?

plueschpruem, 2024-11-26:
> 1. Please attach the file used. (or a sample file)
Happens with every file, I attached one I just made.
> 2. when you resize the window back to original size, does it resize the model correctly, or does it shows the same issue?
It is consistent. Window is opened on internal screen.
Size changes are reflected in the window content, but the centering of the scene is always wrong on internal.
Moving the window to screen 2 keeps the content. When I play/resize on screen 2 it updates to the correct centering.
Moving the window back to screen 1 keeps the content again until the next redraw and it is wrong again.
> 3. Does the mouse pan/resize/rotate of the model works?
Ah those _do_ exist in the new simulator! I was wondering. Scrollwheel zooms like in the normal bench view, pan is auto-active when I press SHIFT and move the mouse, same for rotate with ALT. Odd but a solution for now.

[Test1.FCStd.zip](https://github.com/user-attachments/files/17921477/Test1.FCStd.zip)

shaise, 2024-11-27:
@viernullvier ,
Can you please elaborate? I thought that using the retinaScale is supposed to convert from pixels to monitor coord. Is there another way (that is compatible on all platforms, not just MAC)

sslupsky, 2024-12-04:
> I can confirm that the SimulatorGL window is not set up for processing any touch events at all.

Yeah, this is a problem when using a Magic Mouse as well.  I cannot pan or rotate the model in the new simulator window.

luzpaz, 2024-12-04:
@sslupsky can you add your full About info, TIA

sslupsky, 2024-12-05:
> @sslupsky can you add your full About info, TIA

@luzpaz Updated.

JohnOCFII, 2025-04-20:
> [@viernullvier](https://github.com/viernullvier) ,
> Can you please elaborate? I thought that using the retinaScale is supposed to convert from pixels to monitor coord. Is there another way (that is compatible on all platforms, not just MAC)

@shaise - I wonder if DevicePixelRatio() (as used in this PR: https://github.com/FreeCAD/FreeCAD/pull/19929) would resolve this issue.

Just as a reminder -- the new simulator opens up an object looking like this: 

[image]

When ideally it'd open looking like this: 

[image]

shaise, 2025-04-21:
I will take a look this weekend

luzpaz, 2025-05-09:
@shaise @tarman3 any chance you can look into this ?

tarman3, 2025-05-10:
I can not reproduce this

luzpaz, 2025-05-10:
@tarman3 Do you mind specifying your About info ?

tarman3, 2025-05-10:
<details>
  <summary>FreeCAD 1.0.0.39109 x86_64 ArchLinux KDE wayland</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.5; freecad.gears 1.3.0; btl 0.9.9; fasteners 0.5.38; MeshRemodel 1.10.37; Curves 0.6.60; sheetmetal 0.7.22; Defeaturing 1.2.4
```

</details>

<details>
  <summary>FreeCAD 1.1.0dev.41585 x86_64 AppImage ArchLinux KDE xcb</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41585 (Git) AppImage
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/118.293/1
Installed mods: SearchBar 1.4.2 (Disabled); CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.6; FeedsAndSpeeds 0.6.0 (Disabled); freecad.gears 1.3.0; btl 0.9.9; fasteners.backup1746212036.2932668 (Disabled); fasteners.backup1746276216.4455047 (Disabled); fasteners; MeshRemodel 1.10.37; Curves 0.6.61; sheetmetal 0.7.22; SaveAndRestore 0.2.0; fasteners.backup1746212011.6252146 0.5.38 (Disabled); Defeaturing 1.2.6

```

</details>

<details>
  <summary>FreeCAD 1.1.0dev.41614 x86_64 ArchLinux KDE xcb</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41614 (Git)
Build date: 2025/05/10 05:23:13
Build type: Release
Branch: simmove
Hash: 3965d352fabb890d05c14b5363e1ebd71e19bf27
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/118.293/1
Installed mods: SearchBar 1.4.2 (Disabled); CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.6; FeedsAndSpeeds 0.6.0 (Disabled); freecad.gears 1.3.0; btl 0.9.9; fasteners.backup1746212036.2932668 (Disabled); fasteners.backup1746276216.4455047 (Disabled); fasteners; MeshRemodel 1.10.37; Curves 0.6.61; sheetmetal 0.7.22; SaveAndRestore 0.2.0; fasteners.backup1746212011.6252146 0.5.38 (Disabled); Defeaturing 1.2.6

```

</details>

JohnOCFII, 2025-05-10:
> I can not reproduce this

@tarman3 - looking at the notes above, it appears to be a Mac specific or HiDPI specific issue.

shaise, 2025-05-10:
Indeed it is probably Mac/HiDPi issue. Since I do not have either, its hard for me to debug a fix.

shaise, 2025-05-10:
Though I like it less, but we can place the speed indicator on the left, just above the play button.

shaise, 2025-07-08:
Sorry, Have no idea how to handle this without having macOS, or specific system with the issue.

viernullvier, 2025-07-10:
Oh sorry, completely lost track of this… @shaise I'm not at all familiar with modern Qt on HiDPI environments, so I can't really elaborate. I did have some very limited exposure to GTK on macOS though, so so some classes of HiDPI bugs look familiar to me. One of them is the framework trying to abstract away the HiDPI nature of the display device by mapping one logical pixel to multiple physical pixels, combined with the developer trying to do the exact same thing as well in their own display logic, not knowing that the framework already takes care of it.

I'll happily set up a FreeCAD dev environment and look into a fix in the near future - I'm currently still blocked by outdated macOS/Xcode versions and can't estimate yet how difficult it would be to get everything up and running in the first place.