FreeCAD/FreeCAD #18138: 1.0 is completely innoperable in Win LTSC

ThomasMaker, 2024-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD  1.0 is COMPLETELY non-functioning in LTSC

### Full version info

```shell
Older versons work fine in LTSC but 1.0 doesn't even start up, it installs fine but when it comes to running it there is nothing, no loading or anything of any kind.

Granted LTSC users of FreeCAD are probably few and far between but given that LTSC is basically WIN10 and I'm assuming older versions of FreeCAD work as well in that(have never had 10 on any machine, switched directly from 8.1(last 'private' windows version...) to LTSC) as it does in LTSC.

Given that Older versions work fine in LTSC I'm assuming that 1.0 not working is a minor bug/oversight that is fairly easy to fix...
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-26:
@adrianinsaval any idea?

xtemp09, 2024-11-26:
I think it truly amusing to post a bug report without giving any information at all.

luzpaz, 2025-09-05:
No response from OP. Closing.