FreeCAD/FreeCAD #18141: BIM: Wrong IfcOpenShell version reported

semhustej, 2024-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

Type in Python console:

- `import ifcopenshell`
- `print(ifcopenshell.version)`
- `0.0.0` is returned which is not the correct version number

The wrong number is also exported as an information in the IFC file:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

The issue was confirmed by another Windows user: https://forum.freecad.org/viewtopic.php?t=92355

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-27:
Is this windows exclusively ?

yorikvanhavre, 2024-11-27:
Indeed that's a bug in IfcOpenShell 0.8... It's best to report it to https://github.com/IfcOpenShell/IfcOpenShell
But it doesn't really affect FreeCAD...

Andrej730, 2024-11-27:
What build FreeCAD is using? The build on pip has `.version`.
```python
!pip install ifcopenshell
import ifcopenshell
print(ifcopenshell.version) # 0.8.0
```

yorikvanhavre, 2024-11-27:
I'm compiling myself from sources

furgo16, 2025-01-11:
Is this still an issue now that the upstream bug at https://github.com/IfcOpenShell/IfcOpenShell/issues/5811 has been fixed and there's been reportedly a new conda-forge release?

semhustej, 2025-01-11:
I can still reproduce the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39804 (Git) Conda
Build type: Release
Branch: main
Hash: 73f0e165bee7837704d374d0b1da430027e99fdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

Andrej730, 2025-02-17:
Perhaps weekly builds are still using the older conda-forge build?

yorikvanhavre, 2025-02-18:
> Perhaps weekly builds are still using the older conda-forge build?

Yes I think so. Is the conda build to be disconsidered?