FreeCAD/FreeCAD #18149: Material: Insufficient accuracy in specifying density when changing unit systems.

kaiwas, 2024-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The material density value is shown and set in a convenient way only for some unit systems.
When using millimeters, the density value is so small that zero values ​​are specified.
This also applies to the method of setting up the material. 

Demonstration on video.

https://github.com/user-attachments/assets/ff2c5565-4bf4-47f0-85ff-08f7664cbd16

Probably, instead of the value 0.00 kg/mm3, the value 1.8e-06 kg/mm3 is expected

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * Curves 0.6.51
  * OpenTheme 2024.9.1
  * timber
  * PitchedRoof
  * CurvedShapes 1.0.13
  * FreeCAD_SketchArch
  * frame 0.1.1
  * lattice2 1.0.0
  * sheetmetal 0.5.4
```

### Subproject(s) affected?

Material

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-11-27:
This is a problem with how the quantity object displays values and not specific to materials. It uses the built in routines. I'm not sure there's a good solution to this that will work for all unit systems.

I agree that it is a problem though and not just for materials. This will need some thought.

davesrocketshop, 2024-11-27:
@wwmayer thoughts?

kaiwas, 2024-11-27:
> not specific to materials.

I know that this is not a problem of materials, but rather a system of units.

Ideally, I would like to have each system of units set up in more detail. Number of signs and so on.

For example, I build my drawings (buildings) in millimeters. But at the same time, the volume of materials is calculated in cubic meters. Each time I have to convert by dividing by 10^9

And there are many such examples where, within one system of units, other dimensions may be required that do not correspond to the current system of units.

wwmayer, 2024-11-27:
Setting [this](https://github.com/FreeCAD/FreeCAD/blob/fb7ce6405b21566615912b4150f971a596c8464b/src/Base/Quantity.cpp#L61) value to `Default` will fix the problem because for very small numbers the quantity automatically switches to scientific notation.

reneas, 2024-12-09:
maybe it would be nice to be able to adjust the density unit as well? since kg/mm³ will always be a very small number!
for example osmium which is the densest naturally occurring element on Earth has a density of ~22 g/cm³ which translates to 0.000022 kg/mm³ ...
better units would be g/cm³ and kg/m³

davesrocketshop, 2024-12-12:
> maybe it would be nice to be able to adjust the density unit as well? since kg/mm³ will always be a very small number! for example osmium which is the densest naturally occurring element on Earth [...]

The default units are kg/m^3. One of the side effects of automatically scaling is the units adjust as appropriate.

PR #18334 (2024-12-20): Materials: Change display of Quantity values
The default display type of quantity objects is fixed point resulting in insufficient accuracy when changing unit systems, or when the values are small. This fix changes the default format from 'Fixed' to the more apt 'Default' format. This allows the displayed values to scale as appropriate.

Fixes #18149