FreeCAD/FreeCAD #18151: Possible Bug - error rendered STEP model

kokerkov, 2024-11-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

 My client send me a STEP file containing a spring the other day, after I imported it into FreeCAD, it loads very slowly(about 1 min).

The result is also not correct.
[image]

VS result in Autodesk Inventor
[image]

I think it may be caused by:

- bad model
- step format differences between two softwares
- a bug happened in FreeCAD

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: unset/classic/Qt default
Installed mods: 
  * OpenTheme 2024.8.17
  * Render 2024.9.25
  * sheetmetal 0.4.24
  * workfeature
  * workfeature-macro
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[304209.zip](https://github.com/user-attachments/files/17927902/304209.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-27:
Confirm

<details>
  <summary>FreeCAD 1.1.0dev.39315 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51

```
</details>

tarman3, 2024-11-27:
You can try to fix step file be [Varicad-Viewer](https://www.varicad.com/en/home/products/download/)

kokerkov, 2024-11-28:
> You can try to fix step file by [Varicad-Viewer](https://www.varicad.com/en/home/products/download/)

Thank for the info. We dont have to fix it, it is just for reference .

luzpaz, 2024-11-28:
Can someone provide a diagnosis ?

FEA-eng, 2026-03-15:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260311 (Git shallow)
Build date: 2026/03/11 07:18:44
Build type: Release
Branch: (HEAD detached at 03fc08d02)
Hash: 03fc08d02ebbafa0dc7268d8a7f89e17d3b4db87
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-09-23:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```