FreeCAD/FreeCAD #18159: Sketcher: Inconsistency between Sketcher button tooltips and labeling for Shortcuts.

SpeeQz1, 2024-11-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently in the Sketcher whenever you hover over a button it would show near it a tooltip which should display shortcuts if it has any. The problem here is that not all buttons have their shortcuts listed, and for those which have drop down menus there is different behavior where the dropdown options either have the shortcuts listed or not.

The tooltips specifically for the individual buttons can be confirmed to be missing at `FreeCAD-main/src/Mod/Sketcher/Gui/CommandCreateGeo.cpp` with the property `sAccel`:

[image]
[image]
[image]
[image]

---

[image]
[image]

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.31 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * parts_library
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon 1.4.9
  * SearchBar 1.0.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Anything else?

This could also be the case for buttons in the other workbenches so it would be good for this to be investigated too.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jonezzzzz, 2025-01-03:
I looked into this and found two problems.

1.  The **rectangle**, **polygon** and **spline** dropdown menus are not made with the `GroupCommand` -class, unlike the others.
     -> thats why they dont have the shortcuts listed and the tooltips in the dropdown menu contain only the title.

2. The `setup` - method of `GroupCommand` - class doesn't set the shortcut for the group icon (active tool).
    -> thats why the shortcut is not in the tooltip when you hover over the group icon,
         but is in the tooltip when you hover over the command in the dropdown list.
        (**Create regular polygon** having the shortcut is just a bug, and its the same no matter which tool is selected)

I solved the first problem by implementing all of the dropdown menus with `GroupCommand` -class.

For the second problem I added a `setShortcut` - method to` GroupCommand::setup` - method.
    -> Now the group icon tooltip shows the active tools shortcut, but only after the tool is clicked once.

[image]

To make the shortcut shown even when the tool isnt clicked, a hack can be made by adding the first tools shortcut to the 
implementation of the group command.

[image]

Is this a valid solution for the problem?

luzpaz, 2025-01-03:
@wwmayer perhaps you can offer some orientation if this solution makes sense in regard to the FC codebase ?

Jonezzzzz, 2025-01-26:
I made a draft PR #19253 of the solution I proposed.

closed 2025-02-23 by chennes