FreeCAD/FreeCAD #18165: TechDraw: Axonometric dimensions are placed far away from the drawing area by default.

blank-supportgis, 2024-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating an axonometric length dimension for a large model (lengths of 1000+ mm) in the Tech Draw workbench, the annotation is placed very far away from the model and the actual drawing area. The user has to keep zooming out until the dimension becomes barely visible at the outer edge of the screen to drag it back on the sheet.

Steps to Reproduce:

1. Create a body
2. Create a sketch and draw a rectangle of 1000x1000mm.
3. Pad it to 1000mm to form a cube.
4. Create a Tech Draw page and insert a view of the cube with a scale of 1:10.
5. Create an axonometric dimension on any two neighbouring edges.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * btl 0.9.9 (Disabled)
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-28:
[image]

It is not clear, what is wrong.
Please attach file and screenshot.

maxwxyz, 2024-11-28:
@WandererFan FYI

johndoe2323, 2025-03-22:
\Mod\TechDraw\TechDrawTools\CommandAxoLengthDimension.py is using function getSelEdges of TDToolsUtil.py which appears to give coordinates inflated by factor 10. 
Also, distanceDim.X and distanceDim.Y should be calculated according to the angle of the dim line.

johndoe2323, 2025-04-26:
note: error only reproducible when working on larger objects e.g. >1000mm

luzpaz, 2025-05-07:
> \Mod\TechDraw\TechDrawTools\CommandAxoLengthDimension.py is using function getSelEdges of TDToolsUtil.py which appears to give coordinates inflated by factor 10.
Also, distanceDim.X and distanceDim.Y should be calculated according to the angle of the dim line.

@johndoe2323 would you be open to making a PR ?

WandererFan, 2025-10-05:
We have a duplicate issue here.   Issue #23119 is also about the position of the dimension text.

The same proposed solution is mentioned here: 
https://github.com/FreeCAD/FreeCAD/pull/23228#issuecomment-3362848541

I was going to cancel 23228 and create a new PR that included both the @Syres916 fix for the error message and the missing scale in the calculation.  

I don't think you need any special permission to submit a PR(?).  Just push your branch with the revision to your github repo and then create the PR on the main FreeCAD repo.

johndoe2323, 2025-10-05:
@WandererFan 
ok, created a commit/PR for this (Commit 5f5ad12) - not sure what the issue is with the error message described in #23119

WandererFan, 2025-10-07:
@johndoe2323 I see the commit (in your repo?), but there is no PR associated with it.   You need to go to https://github.com/FreeCAD/FreeCAD/pulls and create a PR that references your branch & commit.  

@hyarion has provided this how-to link: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request.  (Thanks, Benjamin!)

If you do a PR for the scale issue, I'll make one for the superfluous error message in the report view.