FreeCAD/FreeCAD #18168: OpenBSD: Segfault when editing per face colors

thyssentishman, 2024-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Whenever I edit the color of a body's face, when closing the task panel by clicking on `Ok` or `Cancel`, FreeCAD crashes.

Steps to reproduce:

1. New document
2. Create Part->Cube
3. Right click on 'Cube' -> 'Set appearance per face'
4. Click 'Ok' or 'Cancel'

Note that it doesn't seem to make a difference whether a face color was changed or not.

<details>
<summary>Backtrace</summary>
<pre>
Reading symbols from /usr/local/bin/FreeCAD...
[New process 307078]
[New process 159978]
[New process 507604]
[New process 194833]
[New process 204082]
[New process 174048]
[New process 554280]
[New process 521848]
[New process 371220]
[New process 531601]
[New process 155586]
[New process 282728]
[New process 163556]
[New process 257137]
[New process 575576]
[New process 142867]
[New process 609696]
[New process 507037]
[New process 491781]
[New process 154615]
[New process 430981]
[New process 166907]
[New process 500622]
[New process 505249]
[New process 612198]
[New process 551129]
[New process 584547]
[New process 496702]
[New process 499817]
[New process 484634]
[New process 303321]
[New process 460400]
[New process 369437]
[New process 244452]
[New process 258078]
[New process 262786]
[New process 582287]
Core was generated by `FreeCAD'.
Program terminated with signal SIGABRT, Aborted.
#0  thrkill () at /tmp/-:2
2       /tmp/-: No such file or directory.
[Current thread is 1 (process 307078)]
(gdb) bt
#0  thrkill () at /tmp/-:2
#1  0xe1687b741bc224a0 in ?? ()
#2  0x00000120ff64687b in _libc_abort () at /usr/src/lib/libc/stdlib/abort.c:51
#3  0x00000120309f053e in abort_message (format=0x120309e5530 "terminating due to %s exception of type %s")
    at /usr/src/gnu/lib/libcxxabi/../../../gnu/llvm/libcxxabi/src/abort_message.cpp:78
#4  0x00000120309f0646 in demangling_terminate_handler () at /usr/src/gnu/lib/libcxxabi/../../../gnu/llvm/libcxxabi/src/cxa_default_handlers.cpp:77
#5  0x00000120309f0d5a in std::__terminate (func=0x0) at /usr/src/gnu/lib/libcxxabi/../../../gnu/llvm/libcxxabi/src/cxa_handlers.cpp:59
#6  0x00000120309f0ccb in std::terminate () at /usr/src/gnu/lib/libcxxabi/../../../gnu/llvm/libcxxabi/src/cxa_handlers.cpp:88
#7  0x00000120a61e0732 in __clang_call_terminate () from /usr/local/lib/freecad/lib/MatGui.so
#8  0x00000120a623a708 in MatGui::MaterialTreeWidget::~MaterialTreeWidget() () from /usr/local/lib/freecad/lib/MatGui.so
#9  0x00000120a623ab33 in MatGui::MaterialTreeWidget::~MaterialTreeWidget() () from /usr/local/lib/freecad/lib/MatGui.so
#10 0x0000012054a85654 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.5.0
#11 0x00000120389896a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.3.0
#12 0x0000012038ac4443 in QGroupBox::~QGroupBox() () from /usr/local/lib/libQt6Widgets.so.3.0
#13 0x0000012054a85654 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.5.0
#14 0x00000120389896a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.3.0
#15 0x0000012051bad0a3 in PartGui::FaceAppearances::~FaceAppearances() () from /usr/local/lib/freecad/lib/PartGui.so
#16 0x0000012054a85654 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.5.0
#17 0x00000120389896a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.3.0
#18 0x000001203d39f9a3 in QSint::TaskGroup::~TaskGroup() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#19 0x0000012054a85654 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.5.0
#20 0x00000120389896a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.3.0
#21 0x000001203d3868b3 in Gui::TaskView::TaskBox::~TaskBox() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#22 0x000001203d38b807 in Gui::TaskView::TaskDialog::~TaskDialog() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#23 0x0000012051baea53 in PartGui::TaskFaceAppearances::~TaskFaceAppearances() () from /usr/local/lib/freecad/lib/PartGui.so
#24 0x000001203d388868 in Gui::TaskView::TaskView::removeDialog() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#25 0x000001203d3893a9 in Gui::TaskView::TaskView::reject() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#26 0x0000012054a925ea in void doActivate<false>(QObject*, int, void**) () from /usr/local/lib/libQt6Core.so.5.0
#27 0x0000012038b3a5e6 in QDialogButtonBoxPrivate::handleButtonClicked() () from /usr/local/lib/libQt6Widgets.so.3.0
#28 0x0000012054a925ea in void doActivate<false>(QObject*, int, void**) () from /usr/local/lib/libQt6Core.so.5.0
#29 0x0000012038a6adb5 in QAbstractButtonPrivate::emitClicked() () from /usr/local/lib/libQt6Widgets.so.3.0
#30 0x0000012038a6abf5 in QAbstractButtonPrivate::click() () from /usr/local/lib/libQt6Widgets.so.3.0
#31 0x0000012038a6c058 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /usr/local/lib/libQt6Widgets.so.3.0
#32 0x000001203899cc59 in QWidget::event(QEvent*) () from /usr/local/lib/libQt6Widgets.so.3.0
#33 0x0000012038b3714b in QPushButton::event(QEvent*) () from /usr/local/lib/libQt6Widgets.so.3.0
#34 0x000001203894292a in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.3.0
#35 0x0000012038944e7d in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.3.0
#36 0x000001203d011103 in Gui::GUIApplication::notify(QObject*, QEvent*) () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#37 0x0000012054a35df6 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/libQt6Core.so.5.0
#38 0x0000012038942fd6 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
   from /usr/local/lib/libQt6Widgets.so.3.0
#39 0x00000120389b6023 in QWidgetWindow::handleMouseEvent(QMouseEvent*) () from /usr/local/lib/libQt6Widgets.so.3.0
#40 0x00000120389b5152 in QWidgetWindow::event(QEvent*) () from /usr/local/lib/libQt6Widgets.so.3.0
#41 0x000001203894292a in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.3.0
#42 0x0000012038943884 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.3.0
#43 0x000001203d011103 in Gui::GUIApplication::notify(QObject*, QEvent*) () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#44 0x0000012054a35df6 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/libQt6Core.so.5.0
#45 0x0000012093e45951 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/local/lib/libQt6Gui.so.7.0
#46 0x0000012093eb1bfb in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Gui.so.7.0
#47 0x00000120a07588e3 in xcbSourceDispatch(_GSource*, int (*)(void*), void*) () from /usr/local/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.4.0
#48 0x0000012017ec458f in g_main_context_dispatch_unlocked () from /usr/local/lib/libglib-2.0.so.4201.13
#49 0x0000012017ec4a33 in g_main_context_iterate_unlocked () from /usr/local/lib/libglib-2.0.so.4201.13
#50 0x0000012017ec4adb in g_main_context_iteration () from /usr/local/lib/libglib-2.0.so.4201.13
#51 0x0000012054cce5f2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Core.so.5.0
#52 0x0000012054a401e5 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Core.so.5.0
#53 0x0000012054a36563 in QCoreApplication::exec() () from /usr/local/lib/libQt6Core.so.5.0
#54 0x000001203cf2fd8d in Gui::Application::runApplication() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#55 0x0000011e0db5a50d in main ()
(gdb) quit
</pre>
</details>

### Full version info

```shell
OS: OpenBSD 7.6 (xcb)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-28:
@davesrocketshop FYI

thyssentishman, 2024-11-28:
It seems that the issue comes from calling the `addRecent` function here:
https://github.com/FreeCAD/FreeCAD/blob/601637320d0392a12fe613cade33f969b0157a6f/src/Mod/Material/Gui/MaterialTreeWidget.cpp#L104-L109

When the user has not selected any materials yet, `m_uuid` is empty causing it to crash FreeCAD when dereferenced here:
https://github.com/FreeCAD/FreeCAD/blob/601637320d0392a12fe613cade33f969b0157a6f/src/Mod/Material/Gui/MaterialTreeWidget.cpp#L487

The following fixed this for me:
```diff
Index: src/Mod/Material/Gui/MaterialTreeWidget.cpp
--- src/Mod/Material/Gui/MaterialTreeWidget.cpp.orig
+++ src/Mod/Material/Gui/MaterialTreeWidget.cpp
@@ -482,6 +482,9 @@ void MaterialTreeWidget::saveRecents()
 
 void MaterialTreeWidget::addRecent(const QString& uuid)
 {
+    if (uuid.isEmpty()) {
+        return;
+    }
     // Ensure it is a material. New, unsaved materials will not be
     try {
         auto material = _materialManager.getMaterial(uuid);

```

Please let me know if this solution is good enough and I'll open a PR.

davesrocketshop, 2024-11-28:
Seems reasonable

luzpaz, 2024-12-03:
> OS: OpenBSD 7.6 (xcb)
> Architecture: x86_64
> Version: 1.0.0.Unknown
> Build type: Release

An aside, @thyssentishman any way we can get openbsd to not show the 'Unknown' string?

xtemp09, 2024-12-03:
@thyssentishman, why is this crash unreproducible on _Windows_?

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

thyssentishman, 2024-12-04:
> > OS: OpenBSD 7.6 (xcb)
> > Architecture: x86_64
> > Version: 1.0.0.Unknown
> > Build type: Release
> 
> An aside, @thyssentishman any way we can get openbsd to not show the 'Unknown' string?

I had a discussion with @wwmayer about this [in the forum](https://forum.freecad.org/viewtopic.php?p=794386#p794386), but I still need to find out how to generate a `Version.h` header...

wwmayer, 2024-12-04:
For some reason an unexpected exception is raised and because it happens inside the destructor the application will be terminated.

> return `_materialMap->at(uuid)` crashes FreeCAD.

Can you figure out what exception is raised by the at() method if not a `std::out_of_range`? According to the standard (https://en.cppreference.com/w/cpp/container/map/at) exactly this exception type should be thrown.

The implementation of getMaterial() is
```
std::shared_ptr<Material> MaterialManager::getMaterial(const QString& uuid) const
{
    try {
        return _materialMap->at(uuid);
    }
    catch (std::out_of_range&) {
        throw MaterialNotFound();
    }
}
```
that throws an new exception MaterialNotFound. This then is handled in addRecent.

xtemp09, 2024-12-04:
> Could you perhaps try to reproduce the below GDB session

Sorry, not on _Windows_.

wwmayer, 2024-12-04:
> I would assume that it is undefined behavior behaving differently across systems. Could you perhaps try to reproduce the below GDB session with the reproductions steps above and see if the reference is indeed empty?

If you abort the dialog without doing anything then uuid is indeed an empty string. But I cannot detect any UB. It would be interesting if you debugged into MaterialManager::getMaterial to see what happens there.

thyssentishman, 2024-12-04:
> Can you figure out what exception is raised by the at() method if not a `std::out_of_range`? According to the standard (https://en.cppreference.com/w/cpp/container/map/at) exactly this exception type should be thrown.

Well it seems that the uncaught exception is of type `Materials::MaterialNotFound` and that the exception of type `std::out_of_range` is correctly handled (see below). Could this be related to #18170?

```
Thread 1 hit Catchpoint 3 (exception thrown), __cxa_throw (thrown_object=0x4e1b9315480, tinfo=0x4e2483f6da8 <typeinfo for std::out_of_range>,
    dest=0x4e2483d19d0 <std::logic_error::~logic_error()>) at /usr/src/gnu/lib/libcxxabi/../../../gnu/llvm/libcxxabi/src/cxa_exception.cpp:258
258         __cxa_eh_globals *globals = __cxa_get_globals();
(gdb) c
Continuing.
 0# Base::Exception::Exception() in /usr/local/lib/freecad/lib/libFreeCADBase.so
 1# Materials::MaterialManager::getMaterial(QString const&) const in /usr/local/lib/freecad/lib/Materials.so
 2# MatGui::MaterialTreeWidget::addRecent(QString const&) in /usr/local/lib/freecad/lib/MatGui.so
 3# MatGui::MaterialTreeWidget::~MaterialTreeWidget() in /usr/local/lib/freecad/lib/MatGui.so
 4# MatGui::MaterialTreeWidget::~MaterialTreeWidget() in /usr/local/lib/freecad/lib/MatGui.so
 5# QObjectPrivate::deleteChildren() in /usr/local/lib/libQt6Core.so.5.0
 6# QWidget::~QWidget() in /usr/local/lib/libQt6Widgets.so.3.0
 7# QGroupBox::~QGroupBox() in /usr/local/lib/libQt6Widgets.so.3.0
 8# QObjectPrivate::deleteChildren() in /usr/local/lib/libQt6Core.so.5.0
 9# QWidget::~QWidget() in /usr/local/lib/libQt6Widgets.so.3.0
10# PartGui::FaceAppearances::~FaceAppearances() in /usr/local/lib/freecad/lib/PartGui.so
11# QObjectPrivate::deleteChildren() in /usr/local/lib/libQt6Core.so.5.0
12# QWidget::~QWidget() in /usr/local/lib/libQt6Widgets.so.3.0
13# QSint::TaskGroup::~TaskGroup() in /usr/local/lib/freecad/lib/libFreeCADGui.so
14# QObjectPrivate::deleteChildren() in /usr/local/lib/libQt6Core.so.5.0
15# QWidget::~QWidget() in /usr/local/lib/libQt6Widgets.so.3.0
16# Gui::TaskView::TaskBox::~TaskBox() in /usr/local/lib/freecad/lib/libFreeCADGui.so
17# Gui::TaskView::TaskDialog::~TaskDialog() in /usr/local/lib/freecad/lib/libFreeCADGui.so
18# PartGui::TaskFaceAppearances::~TaskFaceAppearances() in /usr/local/lib/freecad/lib/PartGui.so
19# Gui::TaskView::TaskView::removeDialog() in /usr/local/lib/freecad/lib/libFreeCADGui.so
20# Gui::TaskView::TaskView::reject() in /usr/local/lib/freecad/lib/libFreeCADGui.so
21# QObject::qt_metacall(QMetaObject::Call, int, void**) in /usr/local/lib/libQt6Core.so.5.0
22# QRadioButton::qt_metacall(QMetaObject::Call, int, void**) in /usr/local/lib/libQt6Widgets.so.3.0
23# QObject::qt_metacall(QMetaObject::Call, int, void**) in /usr/local/lib/libQt6Core.so.5.0
24# QAbstractButton::isCheckable() const in /usr/local/lib/libQt6Widgets.so.3.0
25# QAbstractButton::isCheckable() const in /usr/local/lib/libQt6Widgets.so.3.0
26# QAbstractButton::mouseReleaseEvent(QMouseEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
27# QWidget::event(QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
28# QPushButton::event(QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
29# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
30# QApplication::notify(QObject*, QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
31# Gui::GUIApplication::notify(QObject*, QEvent*) in /usr/local/lib/freecad/lib/libFreeCADGui.so
32# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /usr/local/lib/libQt6Core.so.5.0
33# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /usr/local/lib/libQt6Widgets.so.3.0
34# void QWidgetPrivate::invalidateBackingStore<QRegion>(QRegion const&) in /usr/local/lib/libQt6Widgets.so.3.0
35# void QWidgetPrivate::invalidateBackingStore<QRegion>(QRegion const&) in /usr/local/lib/libQt6Widgets.so.3.0
36# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
37# QApplication::notify(QObject*, QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
38# Gui::GUIApplication::notify(QObject*, QEvent*) in /usr/local/lib/freecad/lib/libFreeCADGui.so
39# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /usr/local/lib/libQt6Core.so.5.0
40# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /usr/local/lib/libQt6Gui.so.7.0
41# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /usr/local/lib/libQt6Gui.so.7.0
42# QXcbConnection::xi2ReportTabletEvent(void const*, QXcbConnection::TabletData*) in /usr/local/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.4.0
43# g_main_context_dispatch in /usr/local/lib/libglib-2.0.so.4201.13
44# g_main_context_pending in /usr/local/lib/libglib-2.0.so.4201.13
45# g_main_context_iteration in /usr/local/lib/libglib-2.0.so.4201.13
46# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /usr/local/lib/libQt6Core.so.5.0
47# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /usr/local/lib/libQt6Core.so.5.0
48# QCoreApplication::exec() in /usr/local/lib/libQt6Core.so.5.0
49# Gui::Application::runApplication() in /usr/local/lib/freecad/lib/libFreeCADGui.so
50# __register_frame_info in /usr/local/bin/FreeCAD
51# typeinfo name for fmt::v10::detail::buffer<int> in /usr/local/bin/FreeCAD
Thread 1 hit Catchpoint 3 (exception thrown), __cxa_throw (thrown_object=0x4e1bd805380, tinfo=0x4e22a78c528 <typeinfo for Materials::MaterialNotFound>,
    dest=0x4e17cb7f930 <Base::Exception::~Exception()>) at /usr/src/gnu/lib/libcxxabi/../../../gnu/llvm/libcxxabi/src/cxa_exception.cpp:258
258         __cxa_eh_globals *globals = __cxa_get_globals();
(gdb) c
Continuing.
libc++abi: terminating due to uncaught exception of type Materials::MaterialNotFound
```

wwmayer, 2024-12-05:
Maybe the linker creates the symbols for the exception classes (in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Material/App/Exceptions.h) twice (in Materials.so and MatGui.so) so that the exception type in the GUI code is different to the type in the App code.

So, you can try to create the file Exceptions.cpp and move the implementations of the constructors there.

closed 2024-12-08 by chennes

thyssentishman, 2025-03-25:
> Maybe the linker creates the symbols for the exception classes (in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Material/App/Exceptions.h) twice (in Materials.so and MatGui.so) so that the [...]

Once again, sorry for the late reply @wwmayer. While the merged patch does avoid the crash, I tried your suggestion above (without the merged patch) and this also solved the issue. Even better, this seems to have fixed #18170 as well! Below are the patches I applied on top of tag:1.0.0, is this what you mean and are they correct?
<details>
<summary>Patches</summary>

```diff
Index: src/Mod/Material/App/CMakeLists.txt
--- src/Mod/Material/App/CMakeLists.txt.orig
+++ src/Mod/Material/App/CMakeLists.txt
@@ -53,6 +53,7 @@ generate_from_xml(ModelPy)
 generate_from_xml(UUIDsPy)
 
 SET(Python_SRCS
+    Exceptions.cpp
     Exceptions.h
     Array2DPy.xml
     Array2DPyImp.cpp
Index: src/Mod/Material/App/Exceptions.cpp
--- src/Mod/Material/App/Exceptions.cpp.orig
+++ src/Mod/Material/App/Exceptions.cpp
@@ -0,0 +1,150 @@
+#include "Exceptions.h"
+
+namespace Materials
+{
+
+Uninitialized::Uninitialized() {}
+
+Uninitialized::Uninitialized(const char* msg) {
+    this->setMessage(msg);
+}
+
+Uninitialized::Uninitialized(const QString& msg) {
+    this->setMessage(msg.toStdString().c_str());
+}
+
+Uninitialized::~Uninitialized() noexcept = default;
+
+ModelNotFound::ModelNotFound() {}
+
+ModelNotFound::ModelNotFound(const char* msg) {
+    this->setMessage(msg);
+}
+
+ModelNotFound::ModelNotFound(const QString& msg) {
+    this->setMessage(msg.toStdString().c_str());
+}
+
+ModelNotFound::~ModelNotFound() noexcept = default;
+
+InvalidMaterialType::InvalidMaterialType() {}
+
+InvalidMaterialType::InvalidMaterialType(const char* msg) {
+    this->setMessage(msg);
+}
+
+InvalidMaterialType::InvalidMaterialType(const QString& msg) {
+    this->setMessage(msg.toStdString().c_str());
+}
+
+InvalidMaterialType::~InvalidMaterialType() noexcept = default;
+
+MaterialNotFound::MaterialNotFound() {}
+
+MaterialNotFound::MaterialNotFound(const char* msg) {
+    this->setMessage(msg);
+}
+
+MaterialNotFound::MaterialNotFound(const QString& msg) {
[... 332 lines cut ...]
 };
 
 class DeleteError: public Base::Exception
 {
 public:
-    DeleteError()
-    {}
-    explicit DeleteError(char* msg)
-    {
-        this->setMessage(msg);
-    }
-    explicit DeleteError(const QString& msg)
-    {
-        this->setMessage(msg.toStdString().c_str());
-    }
-    ~DeleteError() noexcept override = default;
+    DeleteError();
+    explicit DeleteError(char* msg);
+    explicit DeleteError(const QString& msg);
+    ~DeleteError() noexcept override;
 };
 
 }  // namespace Materials
```
</details>

PR #18271 (2024-12-08): Materials: Check if m_uuid is empty before adding to recents
This fixes #18168
PR #23104 (2025-08-25): Split Material Exceptions.h into source and header
This fixes #18168 and #18170. The former issue was worked around with #18271, but it can be reverted now.

Credits go to @wwmayer who initially thought that this could be the issue in https://github.com/FreeCAD/FreeCAD/issues/18168#issuecomment-2521716247.