FreeCAD/FreeCAD #18169: PartDesign: Fillet looses edges after Pad editing - dimension change

Borodacha, 2024-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Part Design when change dimension of Pad or Sketch that defines this pad Fillet losses edges with the error: "invalid edge link" (so it behaves as non parametric)
[image]

### Full version info

```shell
1.0.0 (39109)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-28:
Please, attach file

Borodacha, 2024-11-29:
[iron stand.zip](https://github.com/user-attachments/files/17956127/iron.stand.zip)
For example: editing of any dimensions in initial sketches of Body-Sketch or Body001-Sketch004 immediately lead to the fillet (fillet001 or Fillet003) edge loosing problem

eblanshey, 2024-12-07:
This is a recurring issue for me, as well. Changes to dimensions made in a spreadsheet are causing fillet edges to be lost quite frequently. If I have time to recreate the issue myself I will share a model. But wanted to comment that I'm observing this in the meantime.

NikoKS, 2025-04-21:
Any solution to this problem?

luzpaz, 2025-04-23:
@NikoKS what version can you reproduce on ? Full [About](https://wiki.freecad.org/About) info please

NikoKS, 2025-04-23:
Hello, this is my about info:
OS: macOS 15.1.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.22

In my case, I noticed the bug when doing this:
1. Create a sketch, draw a line
2. Go to sheet metal workbench, Make Base Wall from the line
3. Create a wall on each end of the flat sheet metal
4. Create fillet on the tip of the object
5. Change the bend radius on the previous Bend

Error like this would pop up:
```
12:40:10  <PropertyLinks> PropertyLinks.cpp(429): Unnamed#Fillet.Base auto change element reference Unnamed#Bend ;Edge28;D1aaf;:H:6,E;:H,E.Edge38 -> ;Edge28;D1e06;:H:6,E;:H,E.Edge38
12:40:17  <PropertyLinks> PropertyLinks.cpp(449): Unnamed#Fillet.Base missing element reference Unnamed#Bend ;Edge28;D1e06;:H:6,E;:H,E.Edge38
12:40:19  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;Edge28;D1e06;:H:6,E;:H,E.Edge38
12:40:19  Fillet: Invalid edge link: ;Edge28;D1e06;:H:6,E;:H,E.Edge38
```

FEA-eng, 2025-09-14:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2025-11-14:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251111 (Git shallow)
Build date: 2025/11/11 16:45:10
Build type: Release
Branch: main
Hash: 4fbb389ce1295e04a5ad7f3a63f68b63b4f4036c
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

luzpaz, 2025-11-15:
Is this OCC related ?

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```

2nOrderEDO, 2026-11-09:
I have a similar reproducible case in **FreeCAD 1.1.4**: changing a spreadsheet parameter causes a Part Design fillet to lose its edge references, although the intended edges remain visibly present **and their references are preserved**. Manually reselecting those edges restores the fillet at the same radius, and makes the part work.

**Steps to reproduce**

1. Open the attached model.
2. In `Spreadsheet`, change `enclosure_width` from **165 mm** to **164 mm**.
3. Recompute the document.
4. `Fillet002` fails because its edge references are no longer resolved.
5. Edit `Fillet002` and reselect the intended edges. The fillet works again, but another width change can break the references again.

**Expected behavior**

The fillet should retain its references when changing the width, provided the corresponding edges still exist and the fillet remains geometrically valid.

**Relevant model details**

- `Fillet002` has a radius of **1 mm**.
- Its base is `Pad_bumper_shroud` (internal name `Pad001`), with saved references to `Edge63` and `Edge68`.
- The two preceding Pads use selected subsets of the same sketch and have `Refine` enabled.
- The fillet was previously moved earlier in the Body history. This may make the case related to https://github.com/FreeCAD/FreeCAD/issues/33838, although I have not established whether the underlying cause is the same.

**Report view error**

```text
17:25:31  <PropertyLinks> PropertyLinks.cpp(514): EP_1455_Enclosure_Custom_Bumper#Fillet002.Base missing element reference EP_1455_Enclosure_Custom_Bumper#Pad001 ;#5ff9:1;:H-b63,E.Edge63
17:25:31  <PropertyLinks> PropertyLinks.cpp(514): EP_1455_Enclosure_Custom_Bumper#Fillet002.Base missing element reference EP_1455_Enclosure_Custom_Bumper#Pad001 ;#5ffc:1;:H-b63,E.Edge68
17:25:31  <Exception> FeatureDressUp.cpp(217): Invalid edge link: ;#5ff9:1;:H-b63,E.Edge63
17:25:31  Fillet002: Invalid edge link: ;#5ff9:1;:H-b63,E.Edge63
```

**Full version information**

```text
OS: KDE Flatpak runtime (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.4.45039 (Git)
Build date: 2026/09/28 13:46:34
Build type: Release
Branch: (HEAD detached at 1.1.4)
Hash: 4fd3bf320d9566a27e60069fc8387448aaa3a094
Python 3.13.16, Qt 6.11.3, Coin 4.0.10, Vtk 9.4.2, boost 1_86, Eigen3 3.4.1, PySide 6.11.3
shiboken 6.11.3, SMESH 7.7.1.0, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.933/1
Installed mods: 
  * Fasteners Workbench 0.5.64
  * KicadStepUp Workbench 11.9.6

```

**Attachment:** [EP_1455_Enclosure_Custom_Bumper.zip](https://github.com/user-attachments/files/34588538/EP_1455_Enclosure_Custom_Bumper.zip)

Would this reproducer be useful in this issue, or should I open a separate report?