FreeCAD/FreeCAD #18170: OpenBSD: 'FreeCAD Exception'  when opening Material Editor

thyssentishman, 2024-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to open the material editor I get the following error:
```
<Exception> FreeCAD Exception
```
This happens no matter how I try to open the material editor:

1. Material WB -> Edit...
2. Part Design WB -> Create Body -> Right click Body -> Material -> Launch editor
3. FEM WB -> Material Editor
4. etc.

### Full version info

```shell
OS: OpenBSD 7.6 (xcb)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-28:
@davesrocketshop FYI

wwmayer, 2024-11-30:
In order to find the location where the exception is raised follow the steps below:
1. Open src/Base/exception.cpp in an editor
2. Include the headers `#include <iostream>` and `#include <boost/stacktrace.hpp>`
3. Go to `Exception::Exception()` and add this line `std::cerr << boost::stacktrace::stacktrace();`

Try to reproduce the error. Look at the output of the report window.

thyssentishman, 2024-12-03:
Thanks @wwmayer. Here is what I got:

```
 0# Base::Exception::Exception() in /usr/local/lib/freecad/lib/libFreeCADBase.so
 1# Materials::MaterialNotFound::MaterialNotFound() in /usr/local/lib/freecad/lib/Materials.so
 2# Materials::MaterialManager::getParent(std::__1::shared_ptr<Materials::Material> const&) const in /usr/local/lib/freecad/lib/Materials.so
 3# MatGui::MaterialsEditor::updateMaterialGeneral() in /usr/local/lib/freecad/lib/MatGui.so
 4# MatGui::MaterialsEditor::setMaterialDefaults() in /usr/local/lib/freecad/lib/MatGui.so
 5# MatGui::MaterialsEditor::setup() in /usr/local/lib/freecad/lib/MatGui.so
 6# MatGui::MaterialsEditor::MaterialsEditor(std::__1::shared_ptr<Materials::MaterialFilter>, QWidget*) in /usr/local/lib/freecad/lib/MatGui.so
 7# MatGui::MaterialTreeWidget::editorClicked(bool) in /usr/local/lib/freecad/lib/MatGui.so
 8# QObject::qt_metacall(QMetaObject::Call, int, void**) in /usr/local/lib/libQt6Core.so.5.0
 9# QAbstractButton::isCheckable() const in /usr/local/lib/libQt6Widgets.so.3.0
10# QAbstractButton::isCheckable() const in /usr/local/lib/libQt6Widgets.so.3.0
11# QAbstractButton::mouseReleaseEvent(QMouseEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
12# QWidget::event(QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
13# QPushButton::event(QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
14# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
15# QApplication::notify(QObject*, QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
16# Gui::GUIApplication::notify(QObject*, QEvent*) in /usr/local/lib/freecad/lib/libFreeCADGui.so
17# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /usr/local/lib/libQt6Core.so.5.0
18# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /usr/local/lib/libQt6Widgets.so.3.0
19# void QWidgetPrivate::invalidateBackingStore<QRegion>(QRegion const&) in /usr/local/lib/libQt6Widgets.so.3.0
20# void QWidgetPrivate::invalidateBackingStore<QRegion>(QRegion const&) in /usr/local/lib/libQt6Widgets.so.3.0
21# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
22# QApplication::notify(QObject*, QEvent*) in /usr/local/lib/libQt6Widgets.so.3.0
23# Gui::GUIApplication::notify(QObject*, QEvent*) in /usr/local/lib/freecad/lib/libFreeCADGui.so
24# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /usr/local/lib/libQt6Core.so.5.0
25# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /usr/local/lib/libQt6Gui.so.7.0
26# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /usr/local/lib/libQt6Gui.so.7.0
27# QXcbConnection::xi2ReportTabletEvent(void const*, QXcbConnection::TabletData*) in /usr/local/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.4.0
28# g_main_context_dispatch in /usr/local/lib/libglib-2.0.so.4201.13
29# g_main_context_pending in /usr/local/lib/libglib-2.0.so.4201.13
30# g_main_context_iteration in /usr/local/lib/libglib-2.0.so.4201.13
31# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /usr/local/lib/libQt6Core.so.5.0
32# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /usr/local/lib/libQt6Core.so.5.0
33# QCoreApplication::exec() in /usr/local/lib/libQt6Core.so.5.0
34# Gui::Application::runApplication() in /usr/local/lib/freecad/lib/libFreeCADGui.so
35# __register_frame_info in FreeCAD
36# typeinfo name for fmt::v10::detail::buffer<int> in FreeCAD
17:05:07  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: FreeCAD Exception
```

wwmayer, 2024-12-04:
This is the relevant code:
```
std::shared_ptr<Material>
MaterialManager::getParent(const std::shared_ptr<Material>& material) const
{
    if (material->getParentUUID().isEmpty()) {
        throw MaterialNotFound();
    }

    return getMaterial(material->getParentUUID());
}
```
So, apparently in your case getParentUUID() returns an empty string. It must be checked what is the passed material object.

Besides this, the exception goes through Qt code and is finally handled in GUIApplication::notify. This is not allowed and causes a crash on other platforms like macOS. This means the exception must be handled inside MaterialTreeWidget::editorClicked.

wwmayer, 2024-12-04:
Weird but in https://github.com/FreeCAD/FreeCAD/blob/366fe0b74e6326a91cd46bf903be41df16e405f7/src/Mod/Material/Gui/MaterialsEditor.cpp#L1191 the exception MaterialNotFound is handled.

thyssentishman, 2024-12-05:
Thanks for looking into this @wwmayer. So if I understand correctly, there's two issues?

1. `getParentUUID()` returns an empty string which shouldn't be empty?
2. Just like in #18168, the `MaterialNotFound` exception is *somehow* not handled by `MaterialsEditor::updateMaterialGeneral()` and therefore handled by `GUIApplication::notify` which is not allowed and needs to be worked around as it is done for other platforms like MacOS?

wwmayer, 2024-12-05:
> getParentUUID() returns an empty string which shouldn't be empty?

This I don't know. It would be interesting to know what's the name of the passed material object.

closed 2025-08-25 by chennes