FreeCAD/FreeCAD #18173: valueError and ReferenceError in resetToolController

duub, 2024-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While creating a CAM Job I get this error:

17:42:56  Traceback (most recent call last):
17:42:56    File "C:\Program Files\FreeCAD 1.0\Mod\CAM\Path\Op\Gui\Base.py", line 391, in resetToolController
17:42:56      self.obj.ToolController = tc
17:42:56      ^^^^^^^^^^^^^^^^^^^^^^^
17:42:56  ValueError: {'sclassname': 'class Base::ValueError', 'sErrMsg': 'PropertyLink does not support external object', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'PropertyLink does not support external object', 'btranslatable': False, 'breported': False}

after completing the setup of all the parameters I get this other one:

17:56:56  Traceback (most recent call last):
17:56:56    File "C:\Program Files\FreeCAD 1.0\Mod\CAM\Path\Op\Gui\Base.py", line 391, in resetToolController
17:56:56      self.obj.ToolController = tc
17:56:56      ^^^^^^^^^^^^^^^^^^^^^^^
17:56:57  ReferenceError: Cannot access attribute 'ToolController' of deleted object

If I cancel the creation of a CAM Job and try to create again a job I get this error:

18:02:18  Traceback (most recent call last):
18:02:18    File "C:\Program Files\FreeCAD 1.0\Mod\CAM\Path\Op\Gui\Base.py", line 391, in resetToolController
18:02:18      self.obj.ToolController = tc
18:02:18      ^^^^^^^^^^^^^^^^^^^^^^^
18:02:18  ReferenceError: Cannot access attribute 'ToolController' of deleted object

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Spanish/Spain (es_ES) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.29
  * Manipulator 1.5.7
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

duub, 2024-11-28:
and now I'm getting this:

```
19:57:46  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\CAM\Path\Main\Gui\Job.py", line 1668, in clearSelection
    self.updateSelection()
  File "C:\Program Files\FreeCAD 1.0\Mod\CAM\Path\Main\Gui\Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
```

maxwxyz, 2024-11-29:
PLease add an example file and steps how to reproduce

duub, 2024-11-30:
it's weird, can't reproduce it, some errors I get it once and then can't repeat it

closed 2024-11-30 by duub