FreeCAD/FreeCAD #18179: Inexplicable distortion appears during Fillet

jordancrubin, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Ive included a file to a simple design.  Three fillets of the outer ring work fine.  When adding the 4th fillet, it takes several minutes before it appears with a distortion imposed over the model.

### Full version info

```shell
OS: macOS 14.5
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.25
  * Curves 0.6.42
```

### Subproject(s) affected?

PartDesign

### Anything else?

[flowerpot.FCStd.zip](https://github.com/user-attachments/files/17954192/flowerpot.FCStd.zip)
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-29:
Files contain 4 fillets, but I do not see issue like you show.
Please write step to reproduce.

tarman3, 2024-11-29:
Confirmed

[image]

<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39315 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51

```
</details>

maxwxyz, 2024-11-29:
Known issue, similar to https://github.com/FreeCAD/FreeCAD/issues/11082

closed 2024-11-29 by maxwxyz

maczargit, 2024-11-29:
Workaround for this, is to split the last fillet into two parts.

First fillet bottom and left edge. 
Next fillet right and top.

[image]

tarman3, 2024-11-29:
I was also work around issue by split fillet feature for long and short curves.

[image]

jordancrubin, 2024-11-30:
> I was also work around issue by split fillet feature for long and short curves.
> 
> ![Screenshot_20241129_155928_hor](https://private-user-images.githubusercontent.com/188672424/391107848-bf1c922d-8948-4d8d-800a-8d0d6c64ba3f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5MjY1NjYsIm5iZiI6MTczMjkyNjI2NiwicGF0aCI6Ii8xODg2NzI0MjQvMzkxMTA3ODQ4LWJmMWM5MjJkLTg5NDgtNGQ4ZC04MDBhLThkMGQ2YzY0YmEzZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEzMFQwMDI0MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NDM4YzMxZWI0OTU1YmExNmUxYWU0ZDE0MmMxOGY3ZTNmNWE1OTMxODk4YzFjNGEyMTVlZDQ4MDJkNjVmYWQyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.wzef1UCKGsjBIB3nsitrUyzuNXBbp0PmHl2owCNz4Z8)

Ive tried this but as you see the fillet shape is not the same as the others

[image]

tarman3, 2024-11-30:
What is the difference?

jordancrubin, 2024-12-02:
> What is the difference?

If you look closely. The fillet done all in one has a boxier appearance where the 2step method has a rounded end.

jordancrubin, 2024-12-02:
[image]
Difference close up…

FlachyJoe, 2024-12-02:
A polar pattern works well
[image]

jordancrubin, 2024-12-03:
> A polar pattern works well [...]

I see what you did there, but I feel like were getting into the weeds with these workarounds again.  Consider of course that this is symmetrical and you can get away with that.  More complex shapes would be trouble.

jordancrubin, 2024-12-04:
> Workaround for this, is to split the last fillet into two parts.
> 
> First fillet bottom and left edge. ![obraz](https://private-user-images.githubusercontent.com/3754746/391050637-e6801e9d-c8d8-4aea-a0b8-5dc215fde380.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMyNzk5MjQsIm5iZiI6MTczMzI3OTYyNCwicGF0aCI6Ii8zNzU0NzQ2LzM5MTA1MDYzNy1lNjgwMWU5ZC1jOGQ4LTRhZWEtYTBiOC01ZGMyMTVmZGUzODAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDRUMDIzMzQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGE3N2M4YmIwYzRjZWMxMzkwNjk2ZDkzODZhZjgwZjFiNWVlY2Q1YjA4ZmM4ZWFmZDljNWVmZjRkMzVmMzMyMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.G7RQxy7RdKLWUwopN0rOxsk36EJhNgJvyIPriPyMNdM)
> 
> Next fillet right and top. ![obraz](https://private-user-images.githubusercontent.com/3754746/391050718-2cd25a0d-b2ab-46e7-acc2-1126a45a4d72.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMyNzk5MjQsIm5iZiI6MTczMzI3OTYyNCwicGF0aCI6Ii8zNzU0NzQ2LzM5MTA1MDcxOC0yY2QyNWEwZC1iMmFiLTQ2ZTctYWNjMi0xMTI2YTQ1YTRkNzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDRUMDIzMzQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmRhZTQ5OWYwNGUxNGFmMTMzYjM2OTk1NjBhOGQzNGRlN2NkYzQ2NDJhODRmODE3ZTNlZDYwMGU5NjEyYWIxNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.hUJitlYWVEL0VfPI2O9-uEuluc1W-BbmKUXqO88y0Do)
> 
> ![obraz](https://private-user-images.githubusercontent.com/3754746/391050375-a4285896-67e5-4ed6-a9fa-9ebaee342abf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMyNzk5MjQsIm5iZiI6MTczMzI3OTYyNCwicGF0aCI6Ii8zNzU0NzQ2LzM5MTA1MDM3NS1hNDI4NTg5Ni02N2U1LTRlZDYtYTlmYS05ZWJhZWUzNDJhYmYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDRUMDIzMzQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODc0ZWU5ZTNmNjE0YzFhNmE5MjVhMWE5NmJlMzZkMjM1OGQyMmE5ODNkMzQwMjcwZDIzMDQwYTc5YTg3MWRmNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.yQBWAK80AJ3LcW3xzMJRJUEeEt1lyBgeSYgvLqlCDNA)

You have labeled this as a workaround possible where i've not yet seen an effective workaround demonstrated.