FreeCAD/FreeCAD #18185: BIM: partially initialized module 'bimcommands' has no attribute 'BimReextrude'

sdifru, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I open BIM workbench, it show this error "partially initialized module 'bimcommands' has no attribute 'BimReextrude' (most likely due to a circular import)" and BIM just doesn't work.

### Full version info

```shell
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods:
  * sheetmetal 0.4.26
  * DynamicData 2.67.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.31
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-29:
Can not reproduce.
Please attach video or screenshot.

<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39315 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51

```
</details>

yorikvanhavre, 2024-11-29:
Can't reproduce either. Make sure you don't have the (old) BIM add-on installed?

sdifru, 2024-11-29:
Here is the screenshot:
[image]

> Can't reproduce either. Make sure you don't have the (old) BIM add-on installed?

I don't install the add-on:
[image]

I use [i3 window manager](https://i3wm.org/) if it helps.

tarman3, 2024-11-29:
Try temporary rename folders:
~/.config/FreeCAD
~/.local/share/FreeCAD

sdifru, 2024-11-29:
> Try temporary rename folders: ~/.config/FreeCAD ~/.local/share/FreeCAD

I'm sorry I don't understand what it means.

tarman3, 2024-11-29:
[image]

sdifru, 2024-11-29:
Yeah I mean both ~/.config/FreeCAD and ~/.local/share/FreeCAD exist. What should I rename them into?

tarman3, 2024-11-29:
Rename to any name.
FreeCAD will start like first time with default config.

sdifru, 2024-11-29:
I've done it and it works now. What should I do with the renamed folders now?

closed 2024-11-29 by sdifru