FreeCAD/FreeCAD #18186: when I click on the Part_Design-> subshapeBinder function,it while exit!

become-hero, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the latest version 1.0.0 when I click on the Part_Design-> subshapeBinder function, then Ctrl+c to withdraw, and then click on the subshapeBinder program again will exit directly. and sometime when I start the program again to open the file, the current project is cleared, is there anyone else experiencing this problem.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-29:
Looks like it was resolved in last dev.

maxwxyz, 2024-11-29:
closing then. Please comment if there is still an issue.

closed 2024-11-29 by maxwxyz

become-hero, 2024-12-02:
> closing then. Please comment if there is still an issue.

OK，think you.i am blocked by a firewall.sorry.

become-hero, 2024-12-02:
> Looks like it was resolved in last dev.
> 
>  2024-11-29.11-26-32_00-00.02_0-00.20.mp4 
> Confirmed FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)
> Not Confirmed - FreeCAD 1.1.0dev.39315 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)

wo,this is amazing.ths man.

luzpaz, 2024-12-02:
was it backported ?

maxwxyz, 2024-12-02:
not sure which PR