FreeCAD/FreeCAD #18188: Memory Leak/Crash

realsteve355, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have experienced regular memory leaks/unhandled exceptions etc with 1.0
FreeCAD has corrupted .FCSTD files
I am unable to save
It's a nightmare. 

[image]
[image]
 
I can't add freecad files to this bug. Is that correct? Happy to send files over.
 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/Dark modern/Qt default
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * btl 0.9.9
  * fasteners 0.5.31
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.3.0
  * Help 1.0.3
  * lattice2 1.0.0
  * MnesarcoUtils 0.2.14
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-29:
Packing file to zip

luzpaz, 2024-12-01:
@realsteve355 :point_up_2:

realsteve355, 2024-12-01:
Hello
I think I solved it. Some files I was linking to objects within had not been opened in 1.0. I went thru my files and opened and saved all in 1.0. Seemed to solve the issue.
Steve

closed 2024-12-02 by luzpaz