FreeCAD/FreeCAD #18189: add file support for sh3 files, like used by Sweet Home 3D to be able to use their furniture in BIM

Mannshoch, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are several sh3 files like sh3f which contains furniture.
The Workspace https://github.com/rostskadat/FreeCAD-SH3D is archived.
(I do focus here on furniture files but a general support for Sweet Home 3D would be really cool.)
Please allow to use this furniture in FreeCAD and also save into this file

I propose to:
- Create a library which could access these files (*.sh3f, *.obj) and import it to FreeCAD
- import these files as Tessellated data (and convert them to brep only by request)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39351 (Git) Snap 1210
Build type: Release
Branch: main
Hash: a372678e987a6522e3de4afc7e981ee67cc5cd34
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Anything else?

[Sweet Home 3D - Homepage](http://www.sweethome3d.com/)
[Sweet Home 3D - Online App](https://www.sweethome3d.com/SweetHome3DJSOnline.jsp)
[Sweet Home 3D - Furniture Page](http://www.sweethome3d.com/importModels.jsp)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Mannshoch, 2024-11-29:
Sorry my failure. I only searched in issues and found this old workbench. 

May I should change the subject that way that kind of Library tool should be created? 
As far the pull request seems focusing on import the data.

rostskadat, 2024-11-29:
Hello @Mannshoch,

As you have certainly noticed FreeCAD already knows how to import a Wavefront (.obj) object.
The SH3D furniture library is a zip file containing the a property file serving as an index toward the different resources for a specific furniture (such as its .obj an then its material .mtl file and any other texture image). 

What kind of workflow are you thinking about that is not easily supported and / or cumbersome by the current functionality set?

Thanks for your insight

Mannshoch, 2024-11-29:
I would propose a furniture library like Sweet Home 3D has.
FreeCAD would need a setting where I could set a directory in which I could save downloaded files.
FreeCAD should load that file - may cache it - and provide a possibility to drag and drop the furniture. In FreeCAD it should auto snap on the the next possible floor on the cursor position.

luzpaz, 2024-12-01:
CC @rostskadat