FreeCAD/FreeCAD #18192: BIM: Opening IFC file as Unlocked opens in Strict IFC mode

semhustej, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- Open FreeCAD
- File -> Open -> Select any IFC file
- Select 'Unlocked' in the 'Lock Document' Option
- File opens in strict IFC mode. No non-IFC objects can be created until strict IFC mode is turned off

See behavior in the video:

https://github.com/user-attachments/assets/eb72ea45-c830-4bc5-93d0-1e5fce02e420

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

I was able to reproduce with 

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-12-03:
Indeed that's a strategy we adopted a bit before 1.0, that opening an IFC file would always begin in locked mode. But indeed it contradicts the dialog...

I now changed a bit the policy so it works like this:

* When opening an IFC file, lock/unlock status depends on the setting chosen in the import dialog (or in the preferences)
* When inserting an IFC file in an existing document, it's always in unlocked mode (an IFC document object is created)

semhustej, 2024-12-03:
> * When opening an IFC file, lock/unlock status depends on the setting chosen in the import dialog (or in the preferences)
> * When inserting an IFC file in an existing document, it's always in unlocked mode (an IFC document object is created)

This sound good.

PR #18269 (2024-12-04): BIM: Fixed lock-unlock mode when opening IFC files
Fixes #18192
commit 5d1cba32 (2024-12-04): BIM: Fixed lock-unlock mode when opening IFC files - Fixes #18192
closed 2024-12-04 by yorikvanhavre