FreeCAD/FreeCAD #18193: BIM: Label of saved IFC file in project tree does not change according to the saved file name

semhustej, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- Open empty project
- Switch to strict IFC mode
- File -> Save as -> choose file name "XYZ"
- The label in project tree does not change according to the name of the file, as is customary in FreeCAD
[image]

When the file is Opened in locked IFC mode, the label is displayed correctly.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-29:
@yorikvanhavre not sure if bug or new feature but here is a ping anyway

yorikvanhavre, 2024-12-03:
Indeed because the label takes the name of the IfcProject, not of the IFC file. Not sure if that's something we want to change?

semhustej, 2024-12-03:
> Indeed because the label takes the name of the IfcProject, not of the IFC file. Not sure if that's something we want to change?

I think showing the name of the IfcProject is better way. The thing that confuses me is 

- when I first save the file, the 'Label' Property in freecad is '_IfcProject'.
- When I look at the IFC file in text editor (and BonsaiBIM) 'IfcProject' name is empty"
[image]

- When I close the file and open it in FreeCAD in again locked mode, the 'Label' is changed XYZ.
[image]

- When I save this file again, the name is still not saved to the IFC file

As I am writing this it seems to me that FreeCAD is not saving 'Label' parameter to IfcProject name. When it opens project with no IfcProject name, it defaults to putting File name into the label.

It seems it would be easy to fix (save Label to IfcProejct name), but there is a corner case, when user has project without a IfcProject name and he/she wants for the project to stay that way.

yorikvanhavre, 2024-12-06:
When there is no name for an entity, and since in FreeCAD we cannot leave an object without a name, I decided to use the IFC class instead. But it's prefixed with _ so we know it's a "dummy" name and not an intended name, and we don't save it to the file unless the user changed it. Otherwise after opening a file we'd have many many changed names...

semhustej, 2024-12-06:
- what I find confusing is that when I open XYZ.ifc  (which has no IfcProject Name) file in Strict mode, it's 'Label' property is set to 'XYZ'. Si when I examine the file in FreeCAD, I would think that is the project name, even though it it not. I would expect the label to be '_IfcProject' in that case. 

[XYZ.ifc.zip](https://github.com/user-attachments/files/18041016/XYZ.ifc.zip)

semhustej, 2024-12-06:
I found related behavior, which I think is an issue: 

I am unable to change Project properties.

Steps to reproduce:

- open XYZ.ifc fiel from previous post
- Change 'Label' and 'Long Name' properties to 'A' and 'B'
- Save file
- Close file
- Open file in locked mode
- 'Label' and 'Long Name' properties are not 'A' and 'B', but the original values

See behavior in video:

https://github.com/user-attachments/assets/a4c8a5dd-2660-4807-b6a4-359685086c01

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

yorikvanhavre, 2024-12-20:
> what I find confusing is that when I open XYZ.ifc (which has no IfcProject Name) file in Strict mode, it's 'Label' property is set to 'XYZ'. Si when I examine the file in FreeCAD, I would think that [...]

Indeed there are two concurrent mechanisms here: 1) on file opening, FreeCAD sets the document name (and label) to the file name, 2) BIM tries to set the document label to the project name. Indeed 2) should take precedence here I guess...

For the other issue, indeed changing a document property is not saved to the IFC file. I'll work on a fix.

semhustej, 2024-12-20:
> Indeed there are two concurrent mechanisms here: 1) on file opening, FreeCAD sets the document name (and label) to the file name, 2) BIM tries to set the document label to the project name. Indeed [...]

That would make sense to me. 

Thank you.