FreeCAD/FreeCAD #18194: PartDesign: Feature Request- Round corners on predefined shapes

leoheck, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Is it possible to have a checkbox to create predefined shapes with round corners like what we have for rectangles?

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.5.10
  * Curves 0.6.51
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.4.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-11-29:
Are you thinking about Sketcher shapes or PartDesign primitives?

maxwxyz, 2024-11-29:
If I understand correctly, basically add fillets on all edges?

leoheck, 2024-11-30:
> Are you thinking about Sketcher shapes or PartDesign primitives?

Ah, indeed, I was thinking about the Sletcher.

leoheck, 2024-11-30:
> If I understand correctly, basically add fillets on all edges?

Yes. There are some predefined shapes, like triangle, pentagon, hexagon... then having a checkbox to change the corner style to round would save a lot of time. For a triangle it is already a tedious thing to do.

The rectangle is a special shape than has separated toolbar button. It has this feature already.

Note: @FlachyJoe having the same functionality for the  predefined PartDesign feature could also be useful. However, on my daily use, I probably never used this feature. Maybe just a couple of times since Freecad 0.19

FlachyJoe, 2024-11-30:
Sketcher_CreateFillet could act on all the selected corners. It would be a good workaround because it would be usable on any shape and not only predefined ones.

Primitives are roundable via PartDesign_Fillet and the default checked UseAllEdges. No more is needed IMHO.