FreeCAD/FreeCAD #18195: Clipping displaying the part as a shell

renatoyamane, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When the clipping is applied, the body is displayed in a shell-like form instead of a body
Files for reference are attached.
[test.zip](https://github.com/user-attachments/files/17960984/test.zip)

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-29:
That's how it was intended.
This is the cheapest way to obtain a cross-section

renatoyamane, 2024-11-29:
Hi,
But on the current way it seems the body was created by a revolution of a sketch like this:
[revolution.zip](https://github.com/user-attachments/files/17961273/revolution.zip)

[image]

renatoyamane, 2024-11-29:
Duplicate of this one:
https://github.com/FreeCAD/FreeCAD/issues/5570

closed 2024-11-29 by renatoyamane