FreeCAD/FreeCAD #18196: TechDraw: Area Dimension point matching

Pendelf, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If the centers of the measured elements coincide, then the references in the area measurement also coincide and it is not clear which dimension belongs to which area.

The bottom picture shows a possible variant

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.31
  * PieMenu 1.9.5
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-11-30:
The end of the leader is set to the centre of the face.  As you point out, this might be in a void in the outer face.  I haven't figured out a good algo to pick a different spot yet.

Eventually, the point of the leader should be "draggable" to a new position.

pierreporte, 2025-04-29:
A **150 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.

pierreporte, 2025-05-07:
Per ISO 129-1, the callout must be formatted “t = <value>”.

CStClairMath, 2025-06-08:
> The end of the leader is set to the centre of the face. As you point out, this might be in a void in the outer face. I haven't figured out a good algo to pick a different spot yet.
> 
> Eventually, the point of the leader should be "draggable" to a new position.

I'm new here and learning the ropes so my proposed algorithm may not work for technical reasons - but I would suggest selecting an arbitrary (ideally exterior) edge and insetting a set distance perpendicular from it's midpoint. 

This current issue could potentially still arise if two edges are tangent at their midpoint. If this is easily checked the algorithm should select a different arbitrary edge.

senft-research, 2025-08-10:
> The end of the leader is set to the centre of the face. As you point out, this might be in a void in the outer face. I haven't figured out a good algo to pick a different spot yet.
> 
> Eventually, the point of the leader should be "draggable" to a new position.

Where is this actually handled inside the `TechDraw` mod?

ipatch, 2025-10-06:
would it not make sense to label the lengths to their corresponding shape? something like the below?

[image]

pierreporte, 2025-10-07:
@ipatch No, this is not the standard practice. It will be misleading.

senft-research, 2025-10-07:
@pierreporte what are your thoughts on my potential solution? I am going to have free time in the next coming weeks and could potentially spend some of it executing my solution cleanly

pierreporte, 2025-10-07:
@senft-research honestly I don’t really understand your solution, but if it works I won’t have anything against it. Go ahead and make a PR so everyone could test it.

senft-research, 2025-10-07:
> OK but does that mean that the callout will be attach only on those medial axis? It’s a big limitation.

To be kurt? The issue you are trying to solve is one that probably can only be solved via limitation. 

Yes, it is limiting the point's location on the face... to the only area of the face guaranteed to not overlap with another face. In my humble opinion, it is impossible to prevent overlap, without this restriction.

pierreporte, 2025-10-07:
Are you able to open a PR, even if it’s as a draft? I want to test your solution.

nishendra3, 2026-02-19:
@pierreporte I would like to work on this issue. I would recommend to place the labels where the users click on the area, and allow them to drag it along. I have already a version of draggable origins working.

[image]

pierreporte, 2026-02-19:
@nishendra3 go ahead. Don't hesitate to open a draft PR with your work.

closed 2026-03-28 by kadet1090