FreeCAD/FreeCAD #18202: Sketcher + Blender Controls Panning Issue with Shift Key

JamesMowery, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

### Description of Issue

When attempting to pan around the viewport while continuing to hold down the Shift key (and not releasing it), you cannot release and then re-press and the middle mouse button to quickly pan around the scene again. It currently requires you to release the Shift key, and then press down the Shift key again + Middle Mouse to start a new pan movement.

Issue is reproducible, and another user via Discord was able to reproduce this on a different machine using MacOS (I am using Linux).

### How To Reproduce

- Enable Blender 3D navigation controls
- Create a new document
- Create a sketch on XY plane
- Hold down Shift (and do not release) and then click Middle Mouse to pan, then release middle mouse to complete the pan.
- While Shift is still being held down, Middle Mouse to attempt to pan again. The pan operation will not occur.
- Immediately leaving sketch mode will allow you to pan multiple times without releasing the Shift key, as intended.

### Full version info

```shell
OS: CachyOS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2024.9.1
  * Curves 0.6.51
```

### Subproject(s) affected?

Sketcher

### Anything else?

Confirmation that another user has been able to reproduce this issue:

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-30:
@Rexbas FYI

otede, 2024-12-03:
Same here on Windows 10, stable FC 1.0

closed 2024-12-20 by chennes