FreeCAD/FreeCAD #18205: Part Design: extremely poor pattern performance

eblanshey, 2024-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a part created in part design that uses the polar pattern and mirror to make copies of features. Unfortunately, although the part isn't too complex, the performance is so slow that a single recompute takes 2 minutes. This makes the program unusable slow. Other CAD programs I've used did not have this problem with patterns, so hopefully there is a way to optimize it somehow.

Steps to reproduce:

1. Download my sample model: [Test.zip](https://github.com/user-attachments/files/17963593/Test.zip) . The model contains a simple donut with a cutout that is patterned in a circle, and then mirror a couple of times.
2. Go to the spreadsheet, type something into an empty cell, and press enter.
3. Observe that the recompute takes more than 10 seconds.

Now, the above model is VERY simple for demonstration purposes. I have a slightly more complex model with more complex sketches, and the recompute takes two minutes. Removing the round/chamfer treatments from the pattern cuts the recompute type in half to 1 minute, but it's still slow, which points to a pattern issue.

I am aware that it is possible to disable auto-recompute on a document level, but adding/changing spreadsheet variables frequently makes this a pain to do, as spreadsheets require recomputing to be able to use a new cell's value.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * WebTools 1.0.0
  * Help 1.0.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

eblanshey, 2024-12-04:
@mwganson thanks for looking into this. Isn't the 14MB size a symptom of the problem I'm describing, not the cause? I made the same test model in another proprietary CAD software and the file size was under 3MB. When I tried to resize the cutout window there, recompute time for the whole model was 5 seconds. Not great, but still much better.

As for invalid curves -- is that something I should open another issue for? I'm not sure why that would be happening.

mwganson, 2024-12-04:
Performance is definitely an issue when it comes to models with many subobjects.  TNP mitigation code no doubt plays some role, but I don't think it's the real culprit since even in 0.21 (which did not have the TNP mitigation code) there was a similar performance lag with such models.  I also do not believe the patterning tools themsleves are to blame, which is not to suggest they are already fully optimized.  FreeCAD just struggles at a fundamental level with such models, yours being one of them due to all the subobjects.

There is a macro called Recompute_Profiler that can be useful in debugging where the bottlenecks are when recomputing a model.  Use it in conjunction with make simple copies of objects (part workbench menu -> create a copy -> simple copy).  The simple copies are non-parametric, so the pattern tools are not being called during recomputes.  Use a simple copy as the base for another boolean operation, example as a BaseFeature of a new Body that will be pocketed with one more hole or padded with another feature, etc.  I think you will see the additional feature takes longer to recompute even though the patterning tools are not be used.

mwganson, 2024-12-04:
As to the invalid curve on surface errors, that is probably an upstream OCCT issue related to pocketing curved surfaces.  I could be possible to cure these defects with a tiny fillet or chamfer on the affected edges.  You would want to do the dressup on the initial hole so it can be patterned, I would think.

eblanshey, 2024-12-09:
@mwganson I created another example that I believe provides conclusive evidence that the issue is with the pattern feature, and not just too many sub-objects. I simplified the original model by removing all the mirrors, then I added an extrude on the inside of the donut, and mirrored it to the opposite side. Observe the following (file attached below):

1. Go to the spreadsheet, type something into any of the columns, press enter. The change takes **58s** to recompute.
2. Now delete the last "mirror" feature in the tree.
3. Go to the spreadsheet, type something into any of the columns, press enter. The change takes **7s** to recompute.
4. Now delete the last two features in the tree, the Round and the Pad (the inner extrusion is now gone)
5. Go to the spreadsheet, type something into any of the columns, press enter. The change still takes **7s** to recompute.

In other words: creating the pad on the inside of the donut did NOT increase recompute time, but the _mirror_ of the pad _significantly_ increased recompute time, by 8 times. 

Also note that now the compressed file is only half a megabyte. The FCStd is 4.1MB (but I have compression level = 0 turned on.)

[Test-inner-pad.zip](https://github.com/user-attachments/files/18063690/Test-inner-pad.zip)

mwganson, 2024-12-10:
Nice detective work.  Indeed, the Mirrored002 object is taking up the bulk of the recompute time.

Macro Recompute Profiler output:

18:35:35  Recomputing... (time in seconds, label)
18:35:35  	0.0	Spreadsheet
18:35:35  	0.0009765625	Sketch
18:35:35  	0.0019507408142089844	Sketch001
18:35:35  	0.0029294490814208984	Pad
18:35:35  	0.014734029769897461	Pocket
18:35:35  	0.07600688934326172	Fillet001
18:35:35  	0.06916975975036621	Fillet002
18:35:35  	0.027953147888183594	Pocket001
18:35:35  	0.03187751770019531	PolarPattern001
18:35:35  	3.5015435218811035	PolarPattern
18:35:35  	0.4822371006011963	Pad001
18:35:35  	0.4905421733856201	Fillet
18:35:52  	17.155992031097412	Mirrored002
18:35:52  	0.31951212882995605	Body
18:35:52  	0.0	Part

Clearly, Mirrored002 is taking the bulk of the recomputing time.  Check geometry shows a lot of defects in the model.  These should be fixed in order to eliminate them as a possibility for at least part of the issue.

eblanshey, 2024-12-10:
@mwganson thank you for sharing the Recompute Profiler output. Indeed it shows that the mirror and polar pattern are the culprits. Unfortunately I don't understand FreeCAD well enough to identify the cause of all the errors found using Check Geometry. The two main issues are "Self-intersection found" and "Invalid curve on surface," but it's not clear why.

eblanshey, 2024-12-10:
Removing the first two fillets seems to remove the Check Geometry errors, and the whole operation performs more quickly now. With the mirror: **5s**. Without the mirror: **1.5s**.

So mirror still slows things down, in this case 3x, but the issue compounds when fillets (and errors) are involved. 

I think it still makes sense to keep this issue open, since a 3x or more performance decrease is still a lot. But maybe I should open an issue for the errors that fillets create on this model? Do you have a preference @maxwxyz?

mwganson, 2024-12-10:
After some investigation I found the source of the problem was one of the holes cuts through a seamline, which I worked around by rotating Sketch by 5 degrees (Sketch -> attachment offset angle = 5 degrees).  Then I deleted both of the first 2 fillets, which I think were there as workarounds in the first place.

[image]

Another minor issue is Mirrored002 is not the Body's Tip feature, remedied by making the body active, and in the Mirroed002 context menu select Set tip.

Now the recompute profile numbers are much better:

20:20:41  Recomputing... (time in seconds, label)
20:20:41  	0.0	Spreadsheet
20:20:41  	0.0029282569885253906	Sketch001
20:20:41  	0.0029218196868896484	Sketch
20:20:41  	0.003477334976196289	Pad
20:20:41  	0.01616811752319336	Pocket
20:20:41  	0.014773845672607422	Pocket001
20:20:41  	0.017133712768554688	PolarPattern001
20:20:41  	0.8614988327026367	PolarPattern
20:20:41  	0.12763309478759766	Pad001
20:20:41  	0.18370413780212402	Fillet
20:20:41  	3.335721969604492	Mirrored002
20:20:41  	0.16004252433776855	Body
20:20:41  	0.0	Part

I believe this proves that the errors in the file were chiefly responsible for the magnified performance issue, although the underlying general suboptimal performance remains.

eblanshey, 2024-12-21:
Although errors greatly contribute to reduced performance, further testing and optimization of my models shows that patterns themselves greatly reduce performance because all features added after a pattern feature takes longer to compute.

For example, I have a part that took 2 minutes to recompute (no errors in the model). I moved all the pattern features to be as late in the document tree as possible. Although the resulting part is the same, it greatly reduced computation time.

This can be seen in @mwganson's recompute profiler results above this comment: notice how the pads and pockets take ~0.01s to calculate at first. After the patterns are added, a pad takes ~0.1s, a 10x increase.

My workarounds are:

- Add patterns as late as possible in the tree
- Reduce the number of patterns (if you need to pattern multiple, unrelated features in the same manner, combine them in one pattern)
- Reduce the number of features after a pattern by combining them, if possible (e.g. instead of mirroring a feature, copy or "mirror" sketch geometry within the same sketch across the X or Y axis.)

Unfortunately there is a limit to these workarounds, as some features simply cannot be placed before the pattern. With a few compounding patterns, every following basic feature (pads, pockets) can take 0.5s or more to recompute. Not to mention the long recompute time of the patterns themselves.

bombela, 2025-03-15:
Similar issue trying something that in my mind should be trivial: pattern a flexture type hole to hold batteries (think AA, AAA...).

I did try using a Part::Array of positive shapes to then boolean cut. But it failed by producing invalid geometry. Seemingly because the pattern faces are touching, I can add spacing, and the geometry computes correctly, but that's not the shape I want, as I now have extraneous walls.

I tried instead using a negative shape `((pocket of a sketch with a quarter of the design + sweep to make a chamfer) + mirror + mirror)` doing two linearpattern of it (for X and Y directions) takes so long to compute, I have to terminate freecad.

[image]

PaddleStroke, 2025-04-09:
The file is too complex. To work on this kind of issue we need very minimal examples. Because here it's messy, there are too many variables too many different operations, even a spreadsheet (which may actually be the culprit).

You need to narrow down the issue to one single type of operation with nothing else.