FreeCAD/FreeCAD #18207: Spreadsheet aliases break sometimes when cutting and pasting

SpeeQz1, 2024-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sometimes when using a spreadsheet with cells which have aliases applied to them, if you were to cut and paste the aliases in different positions occasionally the alias may break (even tho it appears unchanged to the user), when it's broken you can't use reference it at all and you will need to manually remove the aliases and name the cells again with them for the error to disappear.

Example error:
```
02:39:15  <Spreadsheet> Cell.cpp(668): Unnamed1#Spreadsheet.B12: Property 'ExampleProperty' not found in 'ExampleProperty'
```

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.31 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * parts_library
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon 1.4.9
  * SearchBar 1.0.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-30:
Can not reproduce.

<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39380 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: ; lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

Please attache file with broken alias.

SpeeQz1, 2024-11-30:
> Can not reproduce.
> FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)
> 
> FreeCAD 1.1.0dev.39380 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)
> 
> Please attache file with broken alias.

I had didn't have an example as I managed to fix the issue long ago by renaming the variables. From what it seems is that if you have several variables, cut them and paste them in different cells, then cut and paste one of the variables and place it back in one of the spots of the previous variables it will cause it to bug out, almost as if the new position of the variable still has some sort of a reference inside of it to a previous one. I believe the error gets fixed upon reloading the document.

[image]

(After messing around a bit by copy pasting)
[image]

I can't give you a faulty file as the error seems to be resolved upon reload.

[Files.zip](https://github.com/user-attachments/files/17964993/Files.zip)

```
11:47:02  <Spreadsheet> Cell.cpp(668): Unnamed#Spreadsheet.B4: Property 'Var2' not found in 'Var2'
11:47:02  Spreadsheet: One or more cells failed contains errors.
11:47:21  <Spreadsheet> Cell.cpp(668): Unnamed#Spreadsheet.B4: Property 'Var1' not found in 'Var1'
11:47:21  Spreadsheet: One or more cells failed contains errors.
```

tarman3, 2024-11-30:
Confirm

<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39380 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: ; lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

https://github.com/user-attachments/assets/588b554f-a38c-4385-8e2c-fc646f939b55

FEA-eng, 2025-11-15:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.20251111 (Git shallow)
Build date: 2025/11/11 16:45:10
Build type: Release
Branch: main
Hash: 4fbb389ce1295e04a5ad7f3a63f68b63b4f4036c
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD/FreeCAD: https://github.com/AstoCAD/FreeCAD/commit/011f001ada1c4671358378c3035b1686841797b1, https://github.com/AstoCAD/FreeCAD/commit/149b529e10f49ca7123745aaf48c2cf2301f46e7. If someone wants to cherry-pick it and create a PR, these are the relevant commits.