FreeCAD/FreeCAD #18209: rotate tool not working

WanderingMaster, 2024-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when rotating after turning off "read only" the object refuses to move

### Full version info

```shell
no errors reported
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WanderingMaster, 2024-11-30:
please reinvent tool or create new transformation tool that works on both objects and sketches

WanderingMaster, 2024-11-30:
the draft>rotate tool

tarman3, 2024-11-30:
Please attach FreeCAD about information, file, screenshot, something else for reproduce issue

tarman3, 2024-11-30:
What should I do to reproduce the problem?

https://github.com/user-attachments/assets/d5fc8e68-af96-4bc1-adea-9f217c72cfbc

Roy-043, 2024-11-30:
I think this is a case where the user tries to manipulate an attched object. The Placement of the object was made read-only for a reason...

IMO this is a  duplicate of #10470.

tarman3, 2024-11-30:
In this situation pop up message <ins>Box cannot be modified because its placement is readonly</ins>
I dont see a problem with this.

WanderingMaster, 2024-12-01:
01:59:49  <Exception> Geometry.cpp(1063): LProp_CLProps::CurvatureNotDefined()
01:59:49  EditModeInformationOverlayCoinConverter: Curvature graph for B-spline with GeoId=2 could not be calculated.
01:59:49  <Exception> Geometry.cpp(1063): LProp_CLProps::CurvatureNotDefined()
01:59:49  EditModeInformationOverlayCoinConverter: Curvature graph for B-spline with GeoId=2 could not be calculated.
00:42:07  <Sketch> SketchObject.cpp(348): Edge too small: Edge3
00:43:11  <Sketch> SketchObject.cpp(348): Edge too small: Edge3
00:43:11  <Exception> syntax error
00:48:03  Sketch155 cannot be modified because its placement is readonly.

WanderingMaster, 2024-12-01:
i repeated this in another file and it worked. in the main file it didn't, after duplicating the sketch I unchecked "read only" yet even though you see it move, it snaps right back to its original position

WanderingMaster, 2024-12-01:
before the update, I had no problem with this.

WanderingMaster, 2024-12-01:
didn't expect people to answer this so fast, I had never used this before, and I disable the report view because it just gets in the way.

tarman3, 2024-12-01:
> i repeated this in another file and it worked. in the main file it didn't,
Please attach this <ins>main</ins> file.

> after duplicating the sketch I unchecked "read only" yet even though you see it move, it snaps right back to its original position

<ins>unchecked "read only"</ins>, How is this done?

WanderingMaster, 2024-12-01:
right click in data tab>status>readonly

WanderingMaster, 2024-12-01:
just tried it with an added pipe. I duplicated it, then tried to rotate it from the center outward. The pipe disappeared and became it's sketch arc while I was rotating, but then once again the rotate didn't work, but the pipe reappeared. did I mention that the duplicate was always still there after the rotate

WanderingMaster, 2024-12-01:
<class 'IndexError'>: list index out of range

WanderingMaster, 2024-12-01:
about to turn in, I'll have to continue this tomorrow night

tarman3, 2024-12-01:
I give up trying to understand how to repeat what you do )))

luzpaz, 2024-12-01:
Thank you for the report. In order to save time we need a few things from you:
* Full About info (pasted) Help > About FreeCAD > Copy to Clipboard
* A succinct example file (zipped and drag/dropped into GitHub textbox of this ticket.
* Screenshot or even better screencast of how you triggered the bug

WanderingMaster, 2024-12-03:
started over in a new file and the problem disappeared, there must have been a corrupted glitch from a complicated piece I was working on

Roy-043, 2024-12-04:
Let's close this then.

closed 2024-12-04 by Roy-043