FreeCAD/FreeCAD #18213: Sketcher constraints have black box backgrounds

depthoffocus, 2024-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Black background blocks have appeared behind sketcher constraints which can make some of the small icons quite difficult/jarring to work with. Apparently regardless of theme choice, and insensitive to the 3D view background colour.

Not sure if a bug or part of an intentional choice. Wasn't in the previous 39315 build. 

These backgrounds could be a really good idea if their colour and padding was configurable and the spacing was right (symbols and text not running up to the edge of the box), if they were rounded slightly, and were not just large blocks when they split over lines etc., but this appears not to be a configurable colour or feature.

Some sort of transparency issue with an overlay?

[image]
[image]

### Full version info

```shell
OS: macOS 14.7.1
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.33
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.51
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-11-30:
@wwmayer looks like an unintended consequence of https://github.com/FreeCAD/FreeCAD/commit/a372678e987a6522e3de4afc7e981ee67cc5cd34 even with _Anti-Aliasing_ set to _None_ the black background squares/rectangles occur which didn't occur normally.

wwmayer, 2024-11-30:
For two and four color components the alpha value isn't considered. That's why it's black when it should be transparent.

closed 2024-12-02 by chennes

DangerToMe, 2025-08-08:
Having same issue with flatpak version 1.0.1 on Linux Mint. It was working fine for days. Then today, this....
[image]
[image]

PR #18217 (2024-12-02): Gui: Do not ignore alpha channel when converting QImage to SoSFImage
Fixes #18213: Sketcher constraints have black box backgrounds
commit 63ad7d3f (2024-12-02): Gui: Do not ignore alpha channel when converting QImage to SoSFImage

Fixes #18213: Sketcher constraints have black box backgrounds