FreeCAD/FreeCAD #18216: FEM: Elmer solver cannot run on windows

farahats9, 2024-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After writing the solver case file and pressing run it doesn't work, I am assuming it is something to do with the solver path in windows, I downloaded the Elmer binaries from [https://www.elmerfem.org/blog/binaries/](https://www.elmerfem.org/blog/binaries/) here is the report output:
```
15:48:08  The FreeCAD standard unit schema mm/kg/s is used. Elmer sif-file writing is however done in SI units.
15:48:08    ElementDimension: 3
15:48:08    C:\Program Files\FreeCAD 1.0\bin/gmsh.exe
15:48:18  Exception in thread Thread-64 (protector):
15:48:18  Traceback (most recent call last):
15:48:18    File "C:\Program Files\FreeCAD 1.0\bin\Lib\threading.py", line 1045, in _bootstrap_inner
15:48:18      self.run()
15:48:18    File "C:\Program Files\FreeCAD 1.0\bin\Lib\threading.py", line 982, in run
15:48:18      self._target(*self._args, **self._kwargs)
15:48:18    File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femsolver\task.py", line 116, in protector
15:48:18      self.run()
15:48:18    File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femsolver\elmer\tasks.py", line 134, in run
15:48:18      self._process = subprocess.Popen(
15:48:18                      ^^^^^^^^^^^^^^^^^
15:48:18    File "C:\Program Files\FreeCAD 1.0\bin\Lib\subprocess.py", line 1026, in __init__
15:48:18      self._execute_child(args, executable, preexec_fn, close_fds,
15:48:18    File "C:\Program Files\FreeCAD 1.0\bin\Lib\subprocess.py", line 1538, in _execute_child
15:48:18      hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
15:48:18                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
15:48:18  FileNotFoundError: [WinError 2] The system cannot find the file specified
```

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * fasteners 0.5.33
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2024-11-30:
Can you check if the path to the executable is correct in Elmer's preferences?

marioalexis84, 2024-11-30:
@FEA-eng can you reproduce this on Windows?

FEA-eng, 2024-12-05:
> @FEA-eng can you reproduce this on Windows?

No such issue for me.

```
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

farahats9, 2024-12-05:
Could it be the version of Elmer I downloaded? when I tested the non-mpi solver it gave a different error that the result file was not found.

sephamorr, 2025-03-07:
I had this issue -- the error
` File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femsolver\elmer\tasks.py", line 134, in run`
is actually trying to run `mpiexec`, so in my case, it was just because I hadn't restarted freecad since installing MPI. It worked fine once I restarted the app and it picked up the new `PATH`.
Are/were you able to find `mpiexec` in a cmd prompt?

FEA-eng, 2025-09-14:
Currently, the Elmer solver doesn't work at all (#21479) but it should be fixed with upcoming refactoring.

closed 2025-09-25 by luzpaz