FreeCAD/FreeCAD #18218: Material: unable to type * as a unit of new material

semhustej, 2024-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a new material in material editor, I am unable to type in * into units (for instance Pa*s)

Steps to reproduce:

- Open Material editor
- Create new material
- Add new physical model
- Select 'Fluid'
- Type in 10 Pa*s into 'Dynamic Viscosity' value
- Hit enter
- The resulting value is 10 Pa

I am able to reproduce this all the time.

See behavior in video:

https://github.com/user-attachments/assets/b22c07ec-5a0b-4b30-a4f9-35dc02eef66b

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Material

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-01:
@davesrocketshop FYI

davesrocketshop, 2024-12-02:
Verified. Assign to me.

semhustej, 2024-12-02:
@davesrocketshop on related note: is it intended behavior, that when a new physical model is created, all the values have no unit assigned to it? It seems that when user types just the number in, it is unitless number (there is no unit even in the FCmat file). See the question in forum: https://forum.freecad.org/viewtopic.php?t=92462

davesrocketshop, 2024-12-02:
> @davesrocketshop on related note: is it intended behavior, that when a new physical model is created, all the values have no unit assigned to it? It seems that when user types just the number in, it [...]

Default units should be shown. This is an error

semhustej, 2024-12-02:
> >
> 
> Default units should be shown. This is an error

Thanks for confirming. Should I create a new GitHub issue or is this 'double' issue enough?

davesrocketshop, 2024-12-02:
This one should suffice

semhustej, 2024-12-03:
One more thing related to this:

I am able to type in wrong units - such as seconds into density value. This might be also area for improvement.
[image]

luzpaz, 2024-12-03:
> One more thing related to this:

@semhustej Probably worth opening another ticket for this

semhustej, 2024-12-03:
> > One more thing related to this:
> 
> @semhustej Probably worth opening another ticket for this

Reported here: https://github.com/FreeCAD/FreeCAD/issues/18272

davesrocketshop, 2025-09-07:
Still relevant

semhustej, 2026-09-25:
Still valid - the behavior is the same as in this comment: https://github.com/FreeCAD/FreeCAD/issues/18218#issuecomment-4083292167