FreeCAD/FreeCAD #18219: Extremely small Sketcher dimension font fixable by forcing orthographic view

otede, 2024-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See the illegible dimension in top left corner. 
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Sketcher

### Anything else?

The fact that it's fixable by forcing orthographic view doesn't mean it's not a bug. There's no logical explanation to "it's as should be." Switching to orthographic view is therefore only a workaround. There's a plethora of bugs related to font/symbol scaling and rendering.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-09-17:
I wonder what made this dimension so small. I've never had such issues on Windows.

luzpaz, 2025-11-17:
@otede have you been able to reproduce on the development version  ?

otede, 2026-02-17:
I've seen that behavior again with the 1.1RC.

FEA-eng, 2026-02-17:
> I've seen that behavior again with the 1.1RC.

Does it happen randomly or all the time / when you do something reproducible? Is it still on Windows? Can you try in the newest weekly build of 1.2.0?

FEA-eng, 2026-05-19:
See [this](https://github.com/FreeCAD/FreeCAD/issues/18219#issuecomment-3916563570)