FreeCAD/FreeCAD #18223: command save seems not working

angelo953, 2024-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi, 
when I try to save I receive an error message from the OS (Win 11) telling me it was impossible to find the file. Of  course because is a new file.  The same happens when I try to save a already existing file with its name or with a different name; I get a similar message: 

_Si è verificato un problema cercando di salvare il file. Questo può essere dovuto al fatto che alcune delle cartelle genitore non esistono, o non si dispone di autorizzazioni sufficienti, o per altri motivi. Dettagli errore:

"Failed to open file: nottolino epson.FCStd.7dfe06ee-674f-4278-8633-ee02cb51d6ac"

Vuoi salvare il file con un nome diverso?_
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

Before installing v1.0 I had 0.19 and 0.20 installed. then I uninstalled .19 and installed new one.  As I encountered the issue I went back to 0.20 and discovered the issue was present. then I uninstalled all versions, restarted the PC and cleaned registry issues with ccleaner, then  reinstalled V1.0. Unfortunately saving is still impossible

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-02:
I cannot reproduce. In the screenshot it appears that you still have the Sketcher dialog open, does it make a difference if you close the Sketch first?

luzpaz, 2024-12-02:
Where are you installing FreeCAD from ?

angelo953, 2024-12-03:
Hi, installed from freeCad site, installer for Windows.

mwganson, 2024-12-03:
Did you try deleting or renaming the user.cfg file?  Look for macro findConfigFiles if you don't know the location of that file.

angelo953, 2024-12-03:
Thanks for hint.
I renamed user.cfg in user/username/appdata/Roaming/freeCAD as well in
user/username/appdata/Roaming/freeCAD/savedPreferencePack/Backups  but it
did not work: (freecad recreated a new user.cfg file).
Seems (from error message:"Failed to open file: fissaggio angolare
v1.FCStd.f895e349-8dc4-4715-a05f-46b4523e3222") it is looking for a file
named like the original one  "fissaggio angolare v1.FCStd" but with a
longer extension *f895e349-8dc4-4715-a05f-46b4523e3222 *that obviously does
not exist.

mwganson, 2024-12-04:
This is likely related to the automatic backup system where a backup file with the .FCBak extension is created.  My understanding on how this works is when you save your file it creates this temporary file with the unique identifier, which then later gets renamed when all goes according to plan.  There could be some sort of file permission error.  Try running as administrator to see if that makes a difference.  You can also try changing the settings to see if that has any effect: edit menu -> preferences -> general -> document.

It is not necessary to install FreeCAD.  You can simply extract one of the .7z files into a folder in your home directory, e.g. c:\users\yourname\FreeCAD.  Then you can find the executable in the bin folder and create a desktop shortcut or create a link in the taskbar or start menu for it.  When executed from the home user folder there should not be any file permission issues.

angelo953, 2024-12-04:
Hi thanks again for the tip!
I report what happened:
Running as admin: did not work
Changing setting for backup file: seemed to work once, checked that file
was saved, then did not work the second time
running from .exe (your last hint) w/o installation of freeCAD : did not
work
the error message is still the same as follows:

*Failed to open file:
scola-bottiglie-2V6.FCStd.c5ee4725-90ae-4d37-bb13-5715bd420d60*

I tried exporting a mesh and the error is : impossible to find the file
check file name and try again. Quite surprising the system refuses to
create a new file.
[image: image.png]

mwganson, 2024-12-05:
Make sure there are no non-latin characters in the complete file path.  This shouldn't matter, but it might.

Did you try disablling the file backups altogether?  The fact it seemed to work once could be an important clue that you are on the right track.

There is a way in windows to remove all the temporary files as a way of reclaiming disk space.  You could try that to see if it clears out that temporary folder.  Run the chkdsk utility to make sure there are no logical errors on the hard drive.

If you have another drive on the computer other than drive c, for exampe drive d or drive e you can try extracting the .7z file there and running the exe from the bin folder of that dirve.  There shouldn't be any file permission errors on a drive other than drive c.

If reinstalling windows is a viable option you could try that, too.

angelo953, 2024-12-05:
Thanks Mark, You are giving much more than assistance.
Following your last hints:
non latin characters : none are in name(s) of files
disabled backups: I just cleaned all marks in settings general document but
it didn't help
running from external disk (USB 3.0 toshiba canvio) AND saving on same
external disk : *it WORKS!*
running from C: AND saving on external disk: it *WORKS*!

it seems freeCAD1.0 has no permission to write on folders where previous
versions were allowed to. I removed the check mark on folder properties
read only but is not enough.
Then I changed .FCStd file property "open with" to freeCAD.exe: did not
work.
saving under
*C:\Users\amana\freeCAD\FreeCAD_1.0.0-conda-Windows-x86_64-py311\xDRAWINGS*
 (i.e*.* a new folder called xDrawings under the freeCAD folder created
while 'saving with different name') *IS WORKING.*
Moreover I can save under *C:\Users\amana\ *but I cannot under
*C:\Users\amana\Documents
*and all other folders under* /Documents.*
That's all for now. I'm happy to have found a way to save freeCAD .FCstd
files and .stl files.
I hope to find a way to allow me to save in any folder I like to.
Thanks