FreeCAD/FreeCAD #18225: v1.0.0: exception on undo in Assembly workbench; saving causes SIGSEGV

mpictor, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the assembly workbench, I created a distance joint. I then realized I'd made a mistake, so I hit ctrl-z. The undo failed.

Thinking things may work better if I restarted freecad, I saved. This caused freecad to crash.

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * boltsfc 2022.11.5
  * Lithophane
  * OpticsWorkbench 1.0.26
  * FreeGrid 2.1.1
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

Assembly

### Anything else?

seen several times in console before the errors, but probably unrelated:
```
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
```
console output from `undo`:
```
<Exception> invalid container
<App> Transactions.cpp(186): Exception on undo 'Create Distance Joint':invalid container
<Exception> invalid container
<App> Transactions.cpp(186): Exception on undo 'Insert Component':invalid container
Running the Python command 'Assembly_CreateJointDistance' failed:
Traceback (most recent call last):
  File "/tmp/.mount_freecaKAJeKe/usr/Mod/Assembly/CommandCreateJoint.py", line 241, in Activated
    activateJoint(5)
  File "/tmp/.mount_freecaKAJeKe/usr/Mod/Assembly/CommandCreateJoint.py", line 61, in activateJoint
    panel = TaskAssemblyCreateJoint(index)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_freecaKAJeKe/usr/Mod/Assembly/JointObject.py", line 1289, in __init__
    UtilsAssembly.setJointsPickableState(self.doc, False)
  File "/tmp/.mount_freecaKAJeKe/usr/Mod/Assembly/UtilsAssembly.py", line 821, in setJointsPickableState
    if obj.TypeId == "App::FeaturePython" and hasattr(obj, "JointType"):
       ^^^^^^^^^^

'NoneType' object has no attribute 'TypeId'Running the Python command 'Assembly_CreateJointDistance' failed:
Traceback (most recent call last):
  File "/tmp/.mount_freecaKAJeKe/usr/Mod/Assembly/CommandCreateJoint.py", line 241, in Activated
    activateJoint(5)
  File "/tmp/.mount_freecaKAJeKe/usr/Mod/Assembly/CommandCreateJoint.py", line 59, in activateJoint
    JointObject.activeTask.reject()
  File "/tmp/.mount_freecaKAJeKe/usr/Mod/Assembly/JointObject.py", line 1326, in reject
    self.deactivate()
  File "/tmp/.mount_freecaKAJeKe/usr/Mod/Assembly/JointObject.py", line 1348, in deactivate
    self.view.removeEventCallback("SoLocation2Event", self.callbackMove)
                                                      ^^^^^^^^^^^^^^^^^

'TaskAssemblyCreateJoint' object has no attribute 'callbackMove'
```
output after `save`:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib64/libc.so.6(+0x3c480) [0x7f0417452480]
#1  0x7f04192efec0 in typeinfo name for boost::iostreams::detail::direct_streambuf<boost::iostreams::basic_array_source<char>, std::char_traits<char> > from /tmp/.mount_freecaKAJeKe/usr/bin/../lib/libFreeCADApp.so+0
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-01:
Can you please also upload a simple example file for convenient reproducibility ?

tarman3, 2024-12-01:
Also try with last dev version of FreeCAD.
Maybe this problem was already fixed.

mpictor, 2024-12-03:
Sounds like it's already fixed - thanks @tarman3!

closed 2024-12-03 by mpictor

luzpaz, 2024-12-03:
This is a nasty bug, fix should be backported if it isn't already.