FreeCAD/FreeCAD #18227: Draft: Snapping to intersection of orthogonal directions from hold points not possible with default snapping radius

semhustej, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to snap intersection of orthogonal directions from two hold points ('Q' points), which is coincidentally place over existing geometry, it is not possible with the default 8px snapping radius, because the snapping defaults to the end point (even when 'Snap to endpoint' is turned off).

See behavior in the video:

https://github.com/user-attachments/assets/0d0676a3-4072-4014-ab7a-06d6b5d5893f

Sample file:
[snapping-sample-file.FCStd.zip](https://github.com/user-attachments/files/17968527/snapping-sample-file.FCStd.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Anything else?

The issue has been confirmed by other users, see forum discussion here: https://forum.freecad.org/viewtopic.php?t=92194

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-12-04:
This is similar to #18228 in that hovered edges currently have snap priority.

semhustej, 2024-12-04:
> This is similar to #18228 in that hovered edges currently have snap priority.

This issue seems similar, but behaves different than https://github.com/FreeCAD/FreeCAD/issues/18228 , because it can be fixed by setting the different radius. There is also workaround of disabling snapping after hold points are created - they remain in the view and user can still track from them.

See video with me following the same steps, the only difference is changing the snap radius form 8px to 20px. When 8px is set, I am unable to snap to orthogonal intersection, when 20px radius is set, I am able to snap to orthogonal intersection:

https://github.com/user-attachments/assets/fb5a6439-4299-4231-b891-c3fc20cffcf2