FreeCAD/FreeCAD #18228: Draft: Snapping to middle between two hold points not possible when the point is over an existing geometry

semhustej, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to snap middle between two hold points ('Q' points) which is coincidentally over an existing geometry, it is not possible, because the snapping defaults to 'Snap to midpoint' of the existing geometry.

Even changing Snapping radius to 20px does not solve this issue, as it does to this related issue: https://github.com/FreeCAD/FreeCAD/issues/18227

See behavior in the video:

https://github.com/user-attachments/assets/c091842f-ca14-4015-ab36-2ba2d04a7fc1

Sample file: 
[snapping-sample-file.FCStd.zip](https://github.com/user-attachments/files/17968537/snapping-sample-file.FCStd.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-12-04:
The key is that you are hovering one of the edges, and therefore snapping to that edge. You can avoid that by carefully placing your cursor. I am not sure if we can change the snap priority here. The snapping radius only affects snapping to the grid.

https://github.com/user-attachments/assets/808b400c-321a-4e82-9c15-448148604152

semhustej, 2024-12-04:
@Roy-043 

I tested this one too, and I the behavior changes with snap radius too as with the other issue. (Not as I originally reported)

This requires quite a bit of fiddeling form the use though. It would be ideal to change the priority of the snap. If that is not possible (or too much work - I would not consider this a priority) or it would break other things, I would at least change the default snap radius. I found only through the help from forum, that this is in fact possible (because I was using the default snap radius).

Note: I see that I can snap to the middle between two points having the 8px snap radius by using the ` key two times. But that is also quite a lot of fiddeling, as one must move the mouse cursor after the ` key is hit.

yorikvanhavre, 2024-12-05:
At some point a "cycling" system had been introduced to allow tou to cycle through the different available snapping solutions... Not sure what happened to this and if it's still working... Could you give it a test, @semhustej ? https://wiki.freecad.org/Draft_Snap

semhustej, 2024-12-05:
> At some point a "cycling" system had been introduced to allow tou to cycle through the different available snapping solutions... Not sure what happened to this and if it's still working... Could you give it a test, @semhustej ? https://wiki.freecad.org/Draft_Snap

Yes, this works, thank you. I had to cycle 2 times to get to the 'middle between two pints'. It is a bit finnicky as you have to move the mouse after hitting the ' key each time, but it is a workable solution. I tried before I reported the issue and was not able to achieve this, but now I am :).

I think that if a user has taken the work to create two hold points, it would not be conterproductive to give snapping related to these hold points a preference. User will seldom create two hold points and then want to snap plain endpoint or midpoint. I understand that it might not be possible or might be too much work for little benefit.

Changing the default snap radius seemed like the easiest fix to help standard user a little bit. I used my experience where I had to change snap radius manually. If snap radius works different way with different themes/systems, than that is not a way to go forward, as too large of a radius might be confusing/counterproductive for other work.

We can close the issue if non of the above is viable. I would not label it as bug, but rather as a feature.

yorikvanhavre, 2024-12-10:
I think what we would need here at some point, is a deeper refactor of the cycling system. 

The snap system basically analyses the shape under the cursor, builds a list of snap solutions, then picks the one that is closest to the cursor (that we call the winner).  At the moment, the cycling feature only switches between the different shapes under the cursor, and takes the winner of the chosen one.

What we would need is to gather all the shapes under the cursor, build all possible solutions, default to the winner, but allow to cycle through all of the found solutions.

This would raise the calculation run while moving the mouse, so there should maybe be some optimization or caching too.