FreeCAD/FreeCAD #18230: Checkbox in Sketcher Elements panel is inactive

macdroid53, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After creating sketch geometry, if you uncheck the edge in Elements panel it toggles hidden.
After creating reference geometry the box is checked and can not be unchecked. Thus the reference can't be toggled hidden.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39393 (Git)
Build type: Release
Branch: main
Hash: a8ecf5c5176573d2a14459783f6ebd872fd3d981
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * DynamicData 2.70.0
  * lattice2 1.0.0
  * sheetmetal 0.5.10
  * Manipulator 1.5.7
  * kicadStepUpMod 11.4.1
  * fasteners 0.5.33
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.3.0
  * ExtremeProDark 2.9.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-01:
Can not reproduce

<details>
  <summary>FreeCAD 1.1.0dev.39380 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: ; lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

Jonezzzzz, 2025-01-06:
External geometry layer cannot be changed. This is due to PR #8944

> Sketcher: Elements Widget - Fix crash when moving external geometry to another layer
> 
> As reported:
> https://forum.freecad.org/viewtopic.php?p=667426#p667426
> 
> Support for moving external geometry to another layer will come in the future.

FEA-eng, 2025-09-14:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2025-11-15:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251111 (Git shallow)
Build date: 2025/11/11 16:45:10
Build type: Release
Branch: main
Hash: 4fbb389ce1295e04a5ad7f3a63f68b63b4f4036c
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

FEA-eng, 2026-08-27:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260825 (Git shallow)
Build date: 2026/08/25 17:50:34
Build type: Release
Branch: (HEAD detached at 964fce2ed)
Hash: 964fce2edc9a7e1905c2fd3fcd6e7be90afc8375
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```