FreeCAD/FreeCAD #18232: Text from Spreadsheet is exported with double quote now

texaldo, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In release 0.21.2 a spreadsheet cell with text, like 'HELLO, was exported in CSV as: HELLO
In release 1.0 it is now exported in CSV as"HELLO" with double quotes!

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

To reproduce : create a spreadsheet whit a text cell, and export it to CSV 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-01:
Can not confirm

[image]

<details>
  <summary>FreeCAD 1.1.0dev.39380 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: ; lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

xtemp09, 2024-12-01:
I cannot reproduce.

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

xtemp09, 2024-12-01:
@texaldo, have you accidentally inserted a delimiter character? This PR #17140 is relevant.

texaldo, 2024-12-03:
Yes, my cell was containing a tab character, so FreeCad 0.21.2 exports it as is , and FreeCad 1.0 exports with double quotes
[SpreasSheetWithTabFCStd.zip](https://github.com/user-attachments/files/17997245/SpreasSheetWithTabFCStd.zip)

xtemp09, 2024-12-03:
@texaldo, this is supposed to happen.