FreeCAD/FreeCAD #18233: BIM: no option of IFC -> add geometric properties in the tree view

semhustej, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As per wiki pages, when IFC file is opened or imported to FreeCAD:

`To edit an object, you must have loaded its geometry properties. This is done either by double-clicking it or right-clicking and choosing IFC -> add geometric properties in the tree view.`

I am not seeing this option in the right-click menu in the IFC group:

[image]

It is possible to add properties by double-clicking

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-12-03:
I just actually found the cause of the problem, in ifc_viewproviders.py, in setupContextMenu(), all new QActions must have `menu` as a third parameter. But feel free to work on it @AbhiramMasna !

AbhiramMasna, 2024-12-04:
@yorikvanhavre I initially thought the issue was related to the element or representation being missing, but after validating those paths, I shifted focus to how the menu and its actions are built dynamically. I noticed that some QAction objects were missing the parent menu, and once I made this explicit, everything worked as expected. This approach not only fixed the problem but also ensured the code is future-proof and follows best practices for PySide and FreeCAD. And i was also i was fascinated on intricate design of FreeCAD's codebase.

closed 2024-12-04 by yorikvanhavre

semhustej, 2024-12-04:
Thanks for fixing the issue @AbhiramMasna ! This is much appreciated.