FreeCAD/FreeCAD #18234: BIM: Unlocked IFC projects preserves IFC objects which were deleted by external application from the IFC file

semhustej, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- Create a unlocked project (A.FCstd), which has an IFC project inside. A slab is a part of this IFC file (B.ifc). Save both of these files.
- Open B.ifc in external application (BonsaiBIM or text editor) and delete the slab. Save the B.ifc file 
- Open A.FCstd in FreeCAD. The slab is still present, even though it has been deleted from B.ifc, which is linked to AFCstd
- Change the 'Shape Mode' of the slab to 'Coin' -> the slab disappears.
- Change the 'Shape Mode' back to 'Shape' -> the slab is not visible. But the slab data is still present.

See behavior in the video with commentary:
https://1drv.ms/v/c/d058e155a9fc1abb/Edeq6l131G9Oga832wpDUGUBAtsXrO0_WGKccFR-flf1Nw?e=0GdDEX

Sample files  (not sure if it will work on someone else's disk though....):

[sample-file.zip](https://github.com/user-attachments/files/17969703/sample-file.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

I can reproduce this behavior with other BIM objects (wall) and different project.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-02:
@yorikvanhavre FYI

yorikvanhavre, 2024-12-03:
Indeed, good catch, that's a problem. There might also be other changes in the IFC data that would require a recomputation (At the moment no check is performed when opening a FCStd file).

We could try checking the whole file, but that might be slow and resource-intensive... Maybe there is a way to know which objects have change since last time, by reading the owner history. I'll have a better look

semhustej, 2024-12-03:
There is also this use case:

- I remove the IFC file (for whatever reason)
- I open FCstd file, it opens up with IFC objects present
- I do delete object 'Wall' from the IFC part of the FCstd file. I get warnings, but the object is deleted
- I save and close the FCstd file
- I return the IFC file to it's original place
- I open the FCstd file. No error warnings. The 'Wall' object is not in FreeCAD, but it is present in the IFC file.

This is a conflicting behavior.

yorikvanhavre, 2024-12-04:
Indeed. If you loose the IFC file, then anything you do in FreeCAD is not tied anymore to it. If you re-attach it later, they will not be in sync anymore.

What should happen is, indeed, on (FreeCAD) file load, each object should be checked against the IFC contents. But that means unfortunately a lot of recomputing, which is a pity because that's exactly what the FreeCAD file format is meant to avoid (there is a cached version of each shape in the file).

I have a few ideas here that could help, but none of those would actually solve the issue for good:

* The ability to embed the IFC file inside the FreeCAD file. That way, we would have a "portable" thing, no risk to loose the IFC part
* Prevent any editing while the IFC file is not there. Only problem, I don't know how to prevent deleting...

semhustej, 2024-12-04:
> * The ability to embed the IFC file inside the FreeCAD file. That way, we would have a "portable" thing, no risk to loose the IFC part

Isn't the hybrid project "portable" in the current state - as it contains the IFC objects inside of the FCstd file? What would happen to the embedded IFC file if user wanted to update it with a new version of the file? Which is quite common practice (e.g. I receive a newer version of the IFC project from architect) My first impression is that this brings more confusion to the system. 

To solve the portability problem, I think a solution akin to the eTransmit tool form other programs seems more practical to me. This command creates a zip file containing all dependent files and reconfigures saved paths to relative, so that when other users unzippes this file, the paths are current and working.

So for the user the structure would be clear:

1. To work with the IFC file as part of the FCstd, IFC file has to be always accessible 
2. If the IFC file is not present, FCstd contains the last content of the file, but you can not edit/delete objects/add to the IFC part of the file in FreeCAD. It can be used just as a reference.
3. To send the project to another user/new location, user has to always take care of sending/moving both files. This would be helped by the eTransmit tool

These are my first thoughts. I think AEC users are well aware and used to working with 'external links'. Not sure if this is the FreeCAD way though, as I have not yet experimented with how links and the assembly workbench work.

semhustej, 2024-12-04:
> * Prevent any editing while the IFC file is not there. Only problem, I don't know how to prevent deleting...

The prevention of not deleting is unfortunate. The synchroinization betwenn IFC - FCstd is proving quite complicated.....

yorikvanhavre, 2024-12-05:
Yes, therefore having an embedded copy of the IFC file might be interesting. It would also -maybe- allow to compare and see if anything has changed...

semhustej, 2024-12-05:
> Yes, therefore having an embedded copy of the IFC file might be interesting. It would also -maybe- allow to compare and see if anything has changed...

That would be certainly a nice feature, to have changes somehow highlighted since the last version.

yorikvanhavre, 2024-12-06:
I'm not sure how far the comparison is easy to achieve, but I'll have a look.