FreeCAD/FreeCAD #18235: TechDraw: Print All Pages gives you a bitmap image

Pendelf, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

You get a bitmap image when you use the “Print All Pages” function

https://github.com/user-attachments/assets/88603405-a41f-4c3c-9f02-585caa71f9df

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.31
  * PieMenu 1.9.5
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pendelf, 2024-12-01:
Screenshot from Inkscape

[image]

maxwxyz, 2024-12-02:
@WandererFan is this the intended behavior or a bug?

WandererFan, 2024-12-02:
Didn't even know there were variations of pdf files.   

From a quick glance, the only difference I see between printPdf and printAllPdf is that printAllPdf sets the resolution of the QPdfWriter and printPdf uses the defaults (pdfWriter.setResolution(printer->resolution());).   No idea if that is relevant.

How do I tell if the pdf is bitmap or vector?

Pendelf, 2024-12-02:
[image]

The files even have different weights. You can check if the file is vector or raster in a vector editor, for example, Inkscape or open pdf in LibreOffice

mpfusion, 2024-12-03:
Just to let you know: I cannot reproduce the issue here. “Export PDF” and “Print All Pages” result in the same PDF both in vector format. And both with same file size.
```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39419 (Git)
Build type: Release
Branch: main
Hash: 7fdc3b807ac3f84ace7f1e42b7c2bc73939ed98c
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.33
  * kicadStepUpMod 11.4.3
```
[export.pdf](https://github.com/user-attachments/files/17993747/export.pdf)
[print_all.pdf](https://github.com/user-attachments/files/17993749/print_all.pdf)

Pendelf, 2024-12-03:
Tried it on Manjaro, everything works great

OS: Manjaro Linux (KDE/plasmawayland/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/Qt default

WandererFan, 2024-12-04:
Just another data point - removing the "setResolution"  from printAllPdf doesn't change the output on linux, still no bit maps

WandererFan, 2024-12-16:
Does anybody know the details of PDF/A-1b? Could printAll and printAllPdf be producing different flavours of Pdf (PDF 1.4 vs PDF/A-1b)?   We just had an issue with transparency due to different flavours of Pdf.

Pendelf, 2025-09-09:
Still relevant

tobiasfalk, 2025-11-12:
Ping

WandererFan, 2025-11-12:
How are you getting a pdf from PrintAllPages?  Via the OS's print-to-pdf option?  

When I compare a file from Export Pdf with a File > Print (OS print-to-pdf) I don't see any differences.  File > Export Pdf output looks the same too.

But. I'm on linux and the OP is on Win.

WandererFan, 2026-04-08:
Note that Mac seems to have Pdf export issues as well: https://forum.freecad.org/viewtopic.php?t=103997

tobiasfalk, 2026-05-03:
@WandererFan 
Interesting that the templates are not rendering correctly, could be connected how SVGs are imported, as I remember correctly the pathe of SVG import and template import works are very similar.
And if this is based on the QT bug in 6.8 it should be fixed by now.
[https://qt-project.atlassian.net/browse/QTBUG-123817](https://qt-project.atlassian.net/browse/QTBUG-123817)