FreeCAD/FreeCAD #18236: Additive pipeline fails with "Tool shape is not valid for boolean operation"

MihaiDiac, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi, I'm trying to create a simple additive pipe using a shape and a curved path which crosses a padded hexagon. This fails with the error "Tool shape is not valid for boolean operation".

If, instead of a curved path, I draw a straight path => it works
If I remove the padded hexagon => it works
If I save the file and open it in Freecad 0.21.2 and try the same operation => it works

I also reproduced the issue in Safe Mode. I'm attaching a picture of what I'm trying to do and the file I'm working on. Thanks!

[image]

[2L-curves.FCStd.zip](https://github.com/user-attachments/files/17970374/2L-curves.FCStd.zip)

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.50
  * fasteners 0.5.31
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-01:
Confirm

<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39380 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: ; lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

tarman3, 2024-12-01:
I deleted arc and create again. Added tangent to vertical axis.
After that additive pipe working fine.

[image]

frelabrosse, 2025-01-05:
I have something similar with the attached file.  Trying an additive pipe with Sketch003 as the profile and SketchSpine as the path, I get `AdditivePipe: Tool shape is not valid for boolean operation` when pressing OK.
[HS65_topPlate_4.FCStd.zip](https://github.com/user-attachments/files/18310568/HS65_topPlate_4.FCStd.zip)

Creating an additive loft using Sketch003 to Sketch006 also fails (`AdditiveLoft: Null shape`).

freecad 1.0.0

frelabrosse, 2025-01-19:
Just to add some information, I just removed everything apart from sketches `Sketch003`, `Sketch004`, `Sketch006` and `SketchSpine` and could then make the additive pipe using all these.  I then copied the result and pasted it into my original file and it works!

NeilCondon, 2025-01-30:
I think I am seeing the same/similar issue..

In the attached file, when I try to construct an `Additive pipe` using the `Seal retainer profile` sketch as the Profile, and `Inlet opening sketch` to define the sweep path, with `Corner Transition == Right Corner`, I get the same error when I press OK:

> Tool shape is not valid for boolean operation

Oddly, FreeCAD does appear to render/preview the Additive Pipe correctly (seen in yellow here):

[image]

...I just can't complete the operation by pressing OK

```OS: macOS 15.1.1
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68```

NeilCondon, 2025-01-30:
[Example.FCStd.zip](https://github.com/user-attachments/files/18604036/Example.FCStd.zip)

I think I am seeing the same/similar issue..

In the attached file, when I try to construct an `Additive pipe` using the `Seal retainer profile` sketch as the Profile, and `Inlet opening sketch` to define the sweep path, with `Corner Transition == Right Corner`, I get the same error when I press OK:

> Tool shape is not valid for boolean operation

Oddly, FreeCAD does appear to render/preview the Additive Pipe correctly (seen in yellow here):

[image]

...I just can't complete the operation by pressing OK

```OS: macOS 15.1.1
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68```

TudbuT, 2025-02-03:
I get a similar issue: I cannot make a subtractive helix to make a thread like this:

[image]

nwalk7800, 2025-03-12:
I was having the same problem as @TudbuT and found that shrinking my sketch width by a hundredth of a mm resolved it.  I think the problem is that a 2.5mm sketch with a 2.5mm pitch overlap in the part.

paranoidi, 2025-05-14:
[additive_pipe.zip](https://github.com/user-attachments/files/20203610/additive_pipe.zip)

Similar issue, attached screenshot and a project file in a zip.

FEA-eng, 2026-02-21:
The original case works for me, just with some redundant line:

[image]

[This one](https://github.com/FreeCAD/FreeCAD/issues/18236#issuecomment-2571649716) makes the model disappear (might be a different issue). [This one](https://github.com/FreeCAD/FreeCAD/issues/18236#issuecomment-2624684979) has no errors.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow)
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```