FreeCAD/FreeCAD #18238: G85 command inserted into GRBL output even with --translate_drill option

zoltan-ky, 2024-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The grbl postprocessor outputs a G85 drill code even with the --translate_drill option present.    The G85 (like the 81,82, 83) should also be translated with the --translate_drill option.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD-themes 2024.7.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

LarryWoestman, 2024-12-03:
Technically this is a Feature Request, not a bug, since the grbl postprocessor code does not contain any code that attempts to implement G85 handling.  For that matter, the refactored grbl postprocessor code doesn't attempt to implement G85 handling, either.

zoltan-ky, 2024-12-03:
The Drill operation used to output non 8X g-codes with the --translate_drill  grbl postprocessor option.  When I export the same Drill operation with FreeCAD version 1.0.0 instead of using the same 8X codes and their translation logic as before now it outputs a new (was not in FreeCAD version 0.22) G85 ('bore') code cycle.   So it breaks previous behaviour therefore I consider that a bug.  It certainly stopped my workflow after upgrading to 1.0.0 from 0.22.   I could be guilty of not checking the 1.0.0 release notes in detail enough - do user's GRBL controllers now all have to be updated to handle the new G85?

Is it possible that  the G85 code was added recently, possibly for some other feature and it got inadvertently introduced into the basic Drill operation as well as a 'good idea'?

zoltan-ky, 2024-12-04:
I just realized what may have caused the G85:  in my  CAM Drill operation that output the G85 code, the base geometry was the 'bottom' Face of an Extruded circle - a cylinder, so this may have activated the job 'Bore-Drill' operation.

If I make the base geometry just a simple 2D Circle Shape, I get the previously expected G8x codes and they _do_ get translated by the --translate_drill argument to the grbl post-processor

So this issue should be converted to a documentation update request/issue - it would need to describe the difference between the two Drill operation types.

LarryWoestman, 2024-12-04:
We still need to add G85 translating ability to the grbl postprocessors so they will work when the drill operation uses a G85.  So both a feature request and a documentation update request.

It is highly likely that other postprocessors and the grbl postprocessors work with G85.  It is only the --translate_drill option that doesn't know how to cope with a G85.

zoltan-ky, 2024-12-04:
Second that.  To clarify, the G85 comes out fine from all the post-processors, but it is the target of the post-processor output file, the CNC controller, that complains about not handling the G85 (I'm tempted to update mine, there's no reason other than cheap marketing that left drill G-codes out from these CNC controllers, but I'm pressed for time), so the `--translate_drill` option is really needed to allow all these CNC controllers to execute code output from FreeCAD Drill operations. So in that sense, I still view it as an omission, bug (G85 not being translated by --translate_drill) because it breaks earlier behaviour.   Was the G85 in any output of FreeCAD 0.22 (I really didn't look in its code, so can't tell at the moment)?  If it was, and I just missed it with my Drill ops,  then "Feature Request" is correct,  but if it wasn't and G85 was introduced since 0.22 without also adding handling to --translate_drill, then it _is_ a BUG.
It's not a problem, either way, I cooked up a workaround so can wait...