FreeCAD/FreeCAD #18240: Sketcher: Pressing Enter without typing anything in the input field for values does nothing

SpeeQz1, 2024-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating something like a rectangle from center point for example in the Sketcher the default values that are present in the field cannot be used if the user presses Enter, all the key does is nothing. Only when a user manually types values can Enter be used. 

[image]

For example pressing `Enter` should set the width to `21.76 mm` and switch to the length input field. Pressing `Enter` again should set the length to `22.27 mm` and be finished with creating the shape.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.31 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * parts_library
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon 1.4.9
  * SearchBar 1.0.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2025-09-04:
I think this issue is no longer relevant.

```ruby
OS: Ubuntu 24.04.3 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42540 (Git)
Build date: 2025/07/20 22:44:18
Build type: Unknown
Branch: (no branch, bisect started on master)
Hash: 6de135872574a91efe2c2338800512ee75481bab
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/unset/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.0584/96.0584/1
```

xtemp09, 2025-09-04:
I think this issue is no longer relevant.

```ruby
OS: Ubuntu 24.04.3 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/96.0584/1
```