FreeCAD/FreeCAD #18242: Blicking screen with transparent material and dark opentheme

X-Ryl669, 2024-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the OpenTheme (from the Addons) and a dark mode, if I set a material to transparent, it make 3D drawing completely instable, blinking, ghosting. It's unusable, you can't see the pointer by itself, it's lost in the multiple pointers trace on screen.

See example below:
[out.webm](https://github.com/user-attachments/assets/e998a544-d6a3-42e3-a5ee-6f6c84e25d18)

I've tried to disable the MSAA antialiasing to no available (I think it's even worse).

### Full version info

```shell
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: French/France (fr_FR)
Installed mods: 
  * lattice2 1.0.0
  * Assembly4 0.50.15
  * OpenTheme 2024.9.1
  * PieMenu 1.9.5
  * kicadStepUpMod 11.4.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-02:
This looks like the ticket needs to be opened at https://github.com/obelisk79/OpenTheme  

cc @obelisk79

X-Ryl669, 2024-12-02:
Tried without OpenTheme and it's the same. My edit is gone since I rebooted sincewhile. Tried all combinations of OpenGL settings (software only, not software only, with and without VBO, with and without antialiasing, with or without double rendering for transparency). I've also tried on both iGPU (Intel) and dGPU (Nvidia with prime-run) and got the exact same results.

I've also tried to disable desktop compositing (but that's not a thing on Wayland anymore) and all effects. Again, same results.

While zooming in the 3D view, I see many ghosts of the previous rendering, it looks like more a transparency rendering issue to me (nothing to do with stylesheet here). It looks like if, for rendering transparent object, the scene was rendered multiple time (without the transparent object and then with the transparent object) and then later composited. It looks like instead of starting with the clear background color (glClear(color)) in the usual OpenGL rendering cycle, it's kind of reusing previous rendering's buffer, leading to that effect (just zoomed in, you see the previous rendering going on).

[image]

luzpaz, 2024-12-02:
Wait...why are you still using RC1 ? 1.0.0 is out

X-Ryl669, 2024-12-02:
Ok, I've now tried with:
```
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * lattice2 1.0.0
  * Assembly4 0.50.15
  * OpenTheme 2024.9.1
  * PieMenu 1.9.5
  * kicadStepUpMod 11.4.1
``` 
and it's still happening as well.

luzpaz, 2024-12-02:
Can you start in Safe Mode and see if you can reproduce?  
https://wiki.freecad.org/Std_RestartInSafeMode

X-Ryl669, 2024-12-03:
It happens as well (except that it's now gray on white).

X-Ryl669, 2024-12-03:
[Test.zip](https://github.com/user-attachments/files/17991431/Test.zip)
Please try with the attached model if it happens for you too. Try to pan around quickly, you'd see the effect.

obelisk79, 2024-12-03:
Based on the behavior reported, I wonder if this is an OpenGL and driver issue.

kwahoo2, 2024-12-17:
> [Test.zip](https://github.com/user-attachments/files/17991431/Test.zip) Please try with the attached model if it happens for you too. Try to pan around quickly, you'd see the effect.

This does seem to be a Wayland issue.

Tried the same compilation (from git, since Appimage always wants use X11/XWayland).
After setting:
`QT_QPA_PLATFORM=xcb bin/FreeCAD`
no issues, works perfecly.

```
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39649 (Git)
Build type: Release
Branch: main
Hash: e458526279e3cf440a8f14f2424e707983de2fc1
Python 3.13.0, Qt 6.8.0, Coin 4.0.1, Vtk 9.2.6, OCC 7.8.2.dev
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Installed mods: 
  * OpticsWorkbench 1.0.26
  * Curves 0.6.52
  * freecad.gears 1.3.0
  * freecad-xr-workbench 0.9.3
  * kicadStepUpMod 11.4.4
```

But running on Wayland
```
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.39649 (Git)
Build type: Release
Branch: main
Hash: e458526279e3cf440a8f14f2424e707983de2fc1
Python 3.13.0, Qt 6.8.0, Coin 4.0.1, Vtk 9.2.6, OCC 7.8.2.dev
```

I observe exactly the same issues you described. 

What interesting I cannot record that ghosting with OBS Studio.

```
glxinfo | grep OpenGL
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon RX 6600 XT (radeonsi, navi23, LLVM 19.1.0, DRM 3.59, 6.12.4-200.fc41.x86_64)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 24.2.8
OpenGL core profile shading language version string: 4.60
```

FEA-eng, 2025-11-12:
> I built FreeCAD from git with Coin from git, and the issue seems to be fixed for me.

So I guess it's time to close the issue.

FEA-eng, 2026-02-11:
See [this](https://github.com/FreeCAD/FreeCAD/issues/18242#issuecomment-3520752991)