FreeCAD/FreeCAD #18247: CAM: No support for the Masso G3 controller

ShamanTcler, 2024-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Masso controllers have a different GCode sequence for tool changes. The "normal" sequence for FreeCAD is "M6 T#" the Masso expects "T$ M6"

I have already written the fix. The code is here: https://github.com/ShamanTcler/FreeCAD_Masso

It is based on the work of Larry Woestman, in his creation of refactored_linuxcnc_post.py. 

I need to figure out how to write the test cases for Python. However, I have used this code to cut chips more than a few times.

### Full version info

```shell
OS: Windows 11 build 22635
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * btl 0.9.9
  * Curves 0.6.50
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

Hopefully the first of many contributions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

LarryWoestman, 2024-12-03:
I reviewed a preliminary version of the changes and they looked reasonable to me.  I would be happy to review a pull request when one is filed.

LarryWoestman, 2025-01-04:
#18845 fixes this issue.