FreeCAD/FreeCAD #18250: Cannot Locally-Edit Splines in Sketcher

dragazo, 2024-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After creating a spline in the sketcher, it can be edited by moving either the control points or the knots. However, in either case, moving one moves literally every point in the entire spline, making it practically impossible to do any kind of local editing of the curve for matching a desired geometry. I think a practical solution is that if you move a control point all the other control points stay in their current positions (and knots will move to accommodate), and similarly if you move a knot all the other knots stay in their current positions (and control points will move to accommodate).

At the very least, it should be possible to do this for moving control points, because I've seen that using `Sketcher_ConstrainBlock` on the other control points allows me to achieve the behavior I want, but with the added annoyance of having to delete the constraint on a point I want to edit then re-add the constraint after moving before I can repeat the process on another point.

### Full version info

```shell
OS: Linux Mint 20 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-03:
@AjinkyaDahale is this solved with your work on B-splines?

AjinkyaDahale, 2024-12-03:
It should be possible to do this for long enough bspline (which means it has a lot of knots and cp's). For a short bspline things are bound to change. Do you mind sharing a file of what you want to model?

dragazo, 2024-12-03:
There does seem to be a behavior where further-down-the-line control points move less, but they all move at least a small amount - presumably there's some kind of decay function being applied. But this type of spline is generated entirely by the control points and knots, so I'm not sure why in the absence of other constraints any of those points should move without me explicitly moving it.

There's nothing specific I'm trying to model currently - I've just been trying to migrate from using onshape to freecad and this is one pain point that I can't seem to get around. The components I usually work on with splines are either decorative or ergonomic (like a logo/icon embossing or the curve of a handle). In either case I try to use as few control points as possible to get the smoothest result, but that amplifies the issue in freecad since neighbor-wise close control points move more when another is moved.

Also just clarifying: I know the curve must change non-locally when a control point/knot is moved, but I don't see why other cps/knots must move.

AjinkyaDahale, 2025-01-04:
> There does seem to be a behavior where further-down-the-line control points move less, but they all move at least a small amount - presumably there's some kind of decay function being applied. But [...]

I think I see what you mean. There is indeed a case where things move when they have no business moving. It also seems this is related to the existence of knot points: if you delete them (note that this only deletes the visual representation of the knot in the B-spline and not the knot itself), the control points do not move. My guess is that the solver wants to move the "visible" points (which, internally, are just other geometries with an "internal alignment" constraint on them) as little as possible. So, it inadvertently considers both moving a little bit as "better" than just the control point staying constant and the knot point moving more.

Moving forward, the picture should be closer to "these points (control points and knot points) always exist but are not give the same priority as physical points, or even as each other".

> There's nothing specific I'm trying to model currently - I've just been trying to migrate from using onshape to freecad and this is one pain point that I can't seem to get around.

Sorry to hear that. I'm currently working on this, but pretty slowly. I'll try to add this improvement a little bit. However, note that this will imply some pieces of the curve moving "more" than they are currently, which is the price of not wanting to move all pieces only a little. There's some "no free lunch" lesson here.
 
> Also just clarifying: I know the curve must change non-locally when a control point/knot is moved, but I don't see why other cps/knots must move.

Could you elaborate what you mean by "non-local" here? In further detail, could you distinguish between "local", "non-local" and "global" here?

dragazo, 2025-01-04:
Sure, for me local would be within the current knot-delimited segment or the two adjacent segments. So basically any local change will have a negligible effect on the curve outside of that area (if any at all, I can't remember the exact math B-splines use).

Non-local and global I would call the same, and would be anything with a non-neglibible effect outside of the +/- 1 knot segment region.

So for instance in onshape, if I move a knot, only the neighboring segments change shape (and a negligible if any amount further away), so moving a point is a local change. In freecad, this would also be the case if not for the fact that moving a knot or control point causes all the others to move as well, which gives rise to a local change at every point, which makes it a global change overall. I think if the moving others issue were resolved it would behave how I expect.

I might have made it confusing in that last quote, cause when I said `I know the curve must change non-locally when a control point/knot is moved` I meant distance-wise proximity. Sorry about that, lol

AjinkyaDahale, 2025-01-05:
@dragazo what you describe is indeed the intention. I'll see if something can be done, but no promises when :D.