FreeCAD/FreeCAD #18252: Assembly - Stuck in transform task

Haxx4realz, 2024-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I´m trying to make an assembly when i got this issue where im stuck at this transform task and i can´t do anything about it. Tried to undo until the object wasnt even imported but it keeps stuck in that task with no option to apply/close.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.68
```

### Subproject(s) affected?

Assembly

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-02:
Can you zip up the .FCStd file and attach it to this ticket?

Haxx4realz, 2024-12-02:
It doesn´t let me save it say Acess Violation.

luzpaz, 2024-12-02:
@PaddleStroke have you seen this before ?

PaddleStroke, 2024-12-03:
That made me thought about the recent https://github.com/FreeCAD/FreeCAD/pull/18001. However looking at the version of the OP (1.0.0.39109) that would PR is not in this. So it cannot be this.

Without ways to replicate it is hard for me to say. Besides I think that this is probably an issue that would be fixed in current main branch by said 18001.

luzpaz, 2024-12-03:
@PaddleStroke maybe it's time for a minor release to fix this bug ?

Haxx4realz, 2024-12-03:
I managed to do it again but can´t understand how, i will try to find the steps to reproduce it, maybe i´m just using the tools in the wrong way.

luzpaz, 2024-12-03:
@Haxx4realz can you try to use the weekly build[^1] from now on and see if you can recreate ?

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

PaddleStroke, 2024-12-03:
@Haxx4realz maybe you're using the tools in the wrong way. But that should not land you in this buggy situation. So if you can find reproduction steps please let us know.

Haxx4realz, 2024-12-09:
ok i just figured out a way to replicate i managed to record it while messing around so you can understand how. Hope it helps !
here´s a link to the video it´s too big to upload here: https://streamable.com/i0wgo7 
and the exact FCStd files i used are in the zip:
[bug.zip](https://github.com/user-attachments/files/18069063/bug.zip)
Let me know if you need anything else

PaddleStroke, 2024-12-11:
The video is not accessible unfortunately. Can you reupload it? Or perhaps it's private

Haxx4realz, 2024-12-11:
Didn´t realized the platform only allowed 2 days, here´s a link to google drive: https://drive.google.com/file/d/1ARwq2UHm4SzPk_T_2xBuF7r_pIMCFzJx/view?usp=drive_link

PaddleStroke, 2024-12-11:
Perhaps on youtube private video then?

luzpaz, 2024-12-11:
> Didn´t realized the platform only allowed 2 days, here´s a link to google drive: https://drive.google.com/file/d/1ARwq2UHm4SzPk_T_2xBuF7r_pIMCFzJx/view?usp=drive_link

Make it public so it doesn't prompt for login to google please?

Haxx4realz, 2024-12-11:
here it is on youtube : https://youtu.be/ut3A20tInXc

PaddleStroke, 2024-12-13:
@Haxx4realz first note is that you are overconstraining your assembly. The parallel joint is not necessary you should make a revolute directly. Same the distance is not necessary you should just use a revolute. This is why it's not moving correctly.

On the bug itself please try to replicate it in latest weekly build.

Haxx4realz, 2024-12-15:
I cannot reproduce it in the latest weekly build, i understand i can use only one revolute, though in my opinion, one distance constraint shouldnt interfere with the revolute one? or the plane constraint with revolut, it is overconstraining but the constraints aren´t opposites, and cause a bunch of red warnings. Guess this bug is unlikely to happen in the future versions then, but i was a complete beginner when i did it, can´t say for sure.

PaddleStroke, 2024-12-16:
I also think the assembly should handle overconstraints better. But this requires more work I guess.
I see lot of new users trying to use distance and parallel joints. But those are actually advanced joints which should mostly never being used. Most assemblies need only fixed/revolute/slider/cylindrical.
We need to think about this @maxwxyz any idea? Perhaps we could add something in the tooltip to discourage users from using ditance.

maxwxyz, 2024-12-16:
not sure, @FreeCAD/design-working-group might have an idea.
So this issue is resolved as the original author cannot reproduce it anymore?