FreeCAD/FreeCAD #18257: Rectangle contained in circle

langrock, 2024-12-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am running into the following two, likely related, problems when placing a rectangle inside a circle within the Sketcher. First, when creating a sketch that contains a large circle and a smaller one fully contained within that outer circle at any position

[image]

and then add a rectangle contained within the smaller circle with their centers coincident and the rectangle's corners on the circle perimeter (another coincidence constraint), padding this sketch results in an incorrect object as shown below.

[image]

Second, extruding the nested circles only and then creating a sketch on top of this object, again placing a rectangle on top of the smaller circle in the same location as before, causes this rectangle to disappear when executing the pad operation. 

I have the suspicion that first problem is due to a zero-width overlap between the circle and rectangle. I tried recreating this in a number of other CAD programs, though, and none of them had problems with this sketch ... One problem with my theory, though, is that even when I make the rectangle large enough to have its corners overlap with the perimeter of the circle, the extruded body is still not correct.

[image]

Any suggestion on what may be going on and how these problems can be avoided would be greatly appreciated.

Thanks,
Carsten

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-03:
All these sketches are not logical for constructing a solid body.

[image]

tarman3, 2024-12-03:
Please, attach screenshot from another CAD.

maxwxyz, 2024-12-03:
Closing this. If there is a bug, please comment and reopen.

closed 2024-12-03 by maxwxyz

langrock, 2024-12-03:
Thanks for the speedy response and clarifications. I agree that this is a corner case, but I was surprised to see the program neither throw errors nor behave as I had expected based on the results seen in other CAD software or even Freecad in slightly simplified scenarios as follows.

I would not have called the initial sketch 'illogical', though, since it is akin to having two rectangles touch each other at the corners, which should form a perfectly acceptable body. And indeed, this is also not a problem in Freecad. Taking this one step further, placing a rectangle inside a circle and extruding it also works.

[image]

The problem only occurs when enclosing this by anther geometry, such as a circle. Enclosing the touching rectangles by a circle results in the expected structure.

[image]