FreeCAD/FreeCAD #18260: TechDraw: Tree View Overlay not Transparent anymore

maxwxyz, 2024-12-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the ArchDetail example file, the tree view is not transparent anymore, only in the TechDraw WB.
When paning or zooming the tree view is transparent again, also when the application window has no focus. But when the FreeCAD window is in focus, the tree view has a the background color of the page background color:
[image]

When moving the page with MMB it is transparent again:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-03:
@WandererFan FYI and @MisterMakerNL (Stylesheets)

WandererFan, 2024-12-04:
Not an overlay user, but I don't recall any changes in TD that might cause this.  

If I was to make a wild guess, I might wonder if the overlay is expecting some behaviour from the current focus widget that is present in the COIN QuarterWidget  but not present in the QGSPage/QGraphicsView widget.   Does it behave for the dependency graph?  That is another QGraphicsView widget.

maxwxyz, 2024-12-04:
@Syres916 no, unfortunately I cannot narrow it down when it stopped working.
@kadet1090 FYI

Syres916, 2024-12-04:
@WandererFan dependency graph behaves correctly as expected in all three builds listed below, I had to force the tree to show as by default it hides with that specific scenario.

As I can't compile on Windows I used the hash of the Windows build to compile the two Linux builds so as much as I could use the same settings/code.

Bad builds:

```
OS: Windows 7 build 7601
Architecture: x86_64
Version: 1.1.0dev.39216 (Git)
Build type: Release
Branch: main
Hash: ed40de3e2390f11b3ce3f5951fc083f954d7d088
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * A2plus 0.4.66
  * CfdOF 1.27.7
  * Curves 0.6.47
  * FreeCAD-themes 2024.9.14
  * freecad.gears 1.2.0
  * sheetmetal 0.4.24
```

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39216 (Git)
Build type: Release
Branch: (HEAD detached at ed40de3e23)
Hash: ed40de3e2390f11b3ce3f5951fc083f954d7d088
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.6.0
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * FreeCAD_SketchArch
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.51
  * Assembly4 0.50.15
  * A2plus 0.4.68
```

Good build:

```
OS: EndeavourOS (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.39216 (Git)
Build type: Release
Branch: (HEAD detached at ed40de3e23)
Hash: ed40de3e2390f11b3ce3f5951fc083f954d7d088
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * freecad.gears 1.3.0
  * FreeCAD-themes 2024.7.24

```

luzpaz, 2024-12-15:
@kadet1090 any chance you could check this out ?

kadet1090, 2024-12-15:
Cannot reproduce, this probably is error due to Qt Version. I don't think that it should be marked as regression as we don't have good known version, so we don't know if it actually ever worked.

luzpaz, 2024-12-15:
@kadet1090 [About](https://wiki.freecad.org/About) info please, TIA

kadet1090, 2024-12-15:
Added to comment

Syres916, 2024-12-15:
> Cannot reproduce, this probably is error due to Qt Version. I don't think that it should be marked as regression as we don't have good known version, so we don't know if it actually ever worked.

I agree this is likely to be Qt version as shown by my _Good_ working build using Qt6.7.2 shown [above](https://github.com/FreeCAD/FreeCAD/issues/18260#issuecomment-2517790334)

maxwxyz, 2024-12-16:
It works in the stable version, so it must have been broken after the release.
Also, the dependencies are the same version (Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1) so maybe this must be a bug introduced in the code itself?

<details><summary>Windows 1.0 Qt5</summary>
<p>
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.3
  * CurvedShapes 1.0.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.1
  * Ship 2024.11.26
  * Silk 0.1.5
```
</p>
</details> 

<details><summary>macOS 1.1dev Qt6</summary>
<p>
```
OS: macOS Ventura (13.7)
Architecture: x86_64
Version: 1.1.0dev.39586 +77 (Git) Conda
Build type: Release
Branch: pr/kadet1090/17564
Hash: e6b3f7de9609d60ad2b6b4af8fdc50d1f794d99d
Python 3.12.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.8.17
  * CfdOF 1.27.5
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.27
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.24
  * Curves 0.6.45
```
</p>
</details> 

<details><summary>macOS 1.0 Qt5</summary>
<p>
```
OS: macOS 13.7.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.8.17
  * CfdOF 1.27.5
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.27
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.24
  * Curves 0.6.45
```
</p>
</details> 

But not on the Windows dev version

<details><summary>Windows 1.0 Qt5</summary>
<p>

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
``` 

</p>
</details>

luzpaz, 2025-07-27:
I removed regression and added qt6 lable

closed 2025-09-01 by maxwxyz