FreeCAD/FreeCAD #18262: Assigning material without appearance settings overwrides appearance

maxwxyz, 2024-12-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The default material for example has no appearance settings defined:
[image]

If you change the appearance of a body or individual faces and then change the material, the appearance settings are lost / overwritten by the new material, also if the new material has no defined appearance attributes.

To reproduce:

1. Create a new body with an additive cube (to have some geometry)
2. Assign the default material --> appearance changes but default material has no appearance settings
3. Right click the cube and choose random color --> appearance changes
4. Assign the default material again (or any other material which has no appearance set) --> color (appearance) is reset again.

Maybe related: 
- https://github.com/FreeCAD/FreeCAD/issues/14992
- https://github.com/FreeCAD/FreeCAD/issues/11210
- https://github.com/FreeCAD/FreeCAD/issues/15642
- https://github.com/FreeCAD/FreeCAD/issues/12415
- https://github.com/FreeCAD/FreeCAD/issues/15181

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Material

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-03:
@davesrocketshop FYI

furgo16, 2025-11-16:
I happened to stumble on this PR while trying to understand how to use the Material system and its relationship with appearance. I've documented my findings in this section in the wiki: https://wiki.freecad.org/Std_SetMaterial#How_assigning_a_material_affects_appearance - please check for correctness.

> I would expect that if I assign a material with a defined appearance to an object, the would get the appearance of the material even if I fiddled with the object's appearance previously.

@semhustej @davesrocketshop while I don't have a strong opinion on the behavior here, I've specifically documented how to revert a manual override so that the user can go back to the material's appearance if an override was set by mistake: https://wiki.freecad.org/Std_SetMaterial#Reverting_an_appearance_override.

semhustej, 2025-11-16:
>  I've documented my findings in this section in the wiki: https://wiki.freecad.org/Std_SetMaterial#How_assigning_a_material_affects_appearance - please check for correctness.
> 

This is great, thanks a lot! The text seems correct to me, at least how I understand how the appearance works.

closed 2025-11-17 by maxwxyz